tonton81/FlexCAN_T4 #1: Problems with running the example on Teensy 4.0

DeMatzen, 2019-10-23:
I tried to run the example on Teensy 4.0 and read out the sent data with a logic analyzer. The sketch is running without crashing, but I don't see any change on the transmit line. I tried CAN1 to 3, but didn't get any output. What was your testing setup?

Thanks in advance!

tonton81, 2019-10-23:
Usually the transmit line doesnt have any activity if its in bus off state. Make sure you have at least 2 nodes on the network able to talk to each other or at very least acknowledge the receipt of a message, a logic analyzer wont help you with this. 2 nodes minimum and proper bitrate, then you may analyze afterwards with the logic analyzer

tonton81, 2019-10-23:
Also make sure you have the CAN line properly terminated with a resistor, and the pins are not reversed.

tonton81, 2019-12-18:
Did you get it working? I put flexcan_t4 in my car which has dual CAN2.0 busses and both streams are working great

klaasdc, 2020-05-01:
Is it possible to test the code by looping Can RX and Can TX? Trying with Can2 but doesn't generate an interrupt.

klaasdc, 2020-05-01:
Is it possible to put the CAN interface in loopback mode?

tonton81, 2020-05-02:
Yes but not implemented yet, youll need to put the flexcan controller in freeze mode and enable loopback mode manual, then deassert freeze mode.

tonton81, 2020-05-17:
> 
> 
> Is it possible to put the CAN interface in loopback mode?

Loopback mode was added to a testing version on the forum.

closed 2021-01-03 by tonton81