tonton81/FlexCAN_T4 #2: macro "swap" requires 2 arguments, but only 1 given

sepp89117, 2019-12-09:
Hi,
I recently switched from Teensy 3.2 to 4.0. Since the normal FlexCan library did not work I tried the _t4 library with Arduino 1.8.10.
I use `FlexCAN_T4 <CAN1, RX_SIZE_256, TX_SIZE_16> Can0;` to have to rebuild as little as possible.
I call in setup:
`Can0.begin ();`
`Can0.setBaudRate (500000);`

When I compile the code, I get the following errors:
c:\program files (x86)\arduino\hardware\tools\arm\arm-none-eabi\include\c++\5.4.1\bits\exception_ptr.h:118:68: error: macro "swap" requires 2 arguments, but only 1 given

         exception_ptr(static_cast<exception_ptr&&>(__o)).swap(*this);

                                                                    ^

c:\program files (x86)\arduino\hardware\tools\arm\arm-none-eabi\include\c++\5.4.1\bits\exception_ptr.h:126:26: error: macro "swap" requires 2 arguments, but only 1 given

       swap(exception_ptr&) _GLIBCXX_USE_NOEXCEPT;

                          ^

c:\program files (x86)\arduino\hardware\tools\arm\arm-none-eabi\include\c++\5.4.1\bits\exception_ptr.h:161:23: error: macro "swap" requires 2 arguments, but only 1 given

     { __lhs.swap(__rhs); }

                       ^

Please tell me what I'm doing wrong.

Thanks and regards
Sebastian

sepp89117, 2019-12-10:
Hello again,
I have recognized in which direction the problem is going. The lib bites itself with the ILI9341_t3 lib. As soon as both libs are defined in a sketch, there are the errors.
Can someone help?

Thanks and regards
Sebastian

tonton81, 2019-12-10:
Please post your sketches so i can try it this week when i get home

sepp89117, 2019-12-10:
Thanks a lot!
I believe that in any code that uses FlexCan_t4, the error can occur.
But here's an example:
```
#include "ILI9341_t3.h"
#include <FlexCAN_T4.h>

FlexCAN_T4<CAN1, RX_SIZE_256, TX_SIZE_16> opcCan;

void setup(void) {
	Serial.begin(115200);

	opcCan.begin();
	opcCan.setBaudRate(canBaud);
}

void loop() {
	
}
```

Best regards
Sebastian

sepp89117, 2019-12-12:
I have solved my problem. I renamed the Macro swap to swapp in the ILI9341_t3 library, as well as all calls. Now it compiles without errors.

Thanks

tonton81, 2019-12-18:
Awesome, thank you

closed 2019-12-18 by tonton81