tonton81/FlexCAN_T4 #3: CANFD

khalm-github, 2019-12-12:
Hi, i am not expert of teensy and i was excited to use its CANFD but i got upset. i set two nodes (TX and RX), i am using MCP2562FD transceiver, i uploaded the example for the two board without any problems, but i got nothing on the can line, also i got nothing on the serial monitor. i am working for 3 days on that and i could not find where is the problem, i review the wiring many times and its look like right, did i forget something ! or someone faced the same issues before !
This is what i see on the serial monitor
[image]

i used this code for receiving
[image]

and this for transmitting
[image]

i followed this diagram for wiring, but i connected  STBY pin of the transceiver just to the ground!
[image]

tonton81, 2019-12-12:
In setup(), you need to define a callback and interrupts:

 FD.enableMBInterrupts();

Then in loop() you need to enable the data buffer:
FD.events();

khalm-github, 2019-12-12:
Thanks you very much for your quick response, but its still no big difference. i would be thankful if you notice me about any other thing may cause this issue .

now my TX code is like that:
[image]

and RX code:

[image]

and on the serial monitor just a white screen

[image]

Is there anything that may be i did wrong !

tonton81, 2019-12-12:
Is your transceiver enabled? If not you need to drive the line LOW to enable it

khalm-github, 2019-12-12:
i will check that again tomorrow because i leaved now then i will back
another hint; what do you mean by drive the line low, i think the UART take control of all pins after serial.begin(), or do you mean something else !

tonton81, 2019-12-12:
The transceiver enable pin, dont know if yours has one or not, didnt have time to check the datasheet of it, but for sure begin() has to be absolutely called first before the rest

khalm-github, 2019-12-12:
no my transciever does not have enable pin. tomorrow i will check again the code, thanks Sir

This is my transciever MCP2562FD from microship

[image]

khalm-github, 2019-12-13:
Hi, i modified the code as you mention, and now i got the old output again.

[image]

i really wondering why it does not work! should i replace the transceiver with another type.

also i disconnect the transceiver and connect the rx and tx pins 30 and 31 for teensy on the oscilloscope and i got signal

tonton81, 2019-12-13:
Check if the transceiver CANH and CANL lines are parallel between eacy Teensy, and that both Teensies share a common ground path

tonton81, 2019-12-18:
Did you get up and running in CANFD mode yet?
I've posted examples of TeensyCAN running CANFD in 8Mbps mode talking between nodes, perhaps you can use that to rule out software issue to check the wiring setup

khalm-github, 2020-01-20:
Hello, sorry for late reply, but i was in a long vacation!

yes the CANFD mode is working now, the STBY pin of the transceiver was not connected, after i connect it to the ground, its worked fine. i test only one teensy. nowadays i am going on test 10 teensy in CANFD mode, after finished, may be i will  back here to give my feedback !

closed 2021-01-03 by tonton81