tonton81/FlexCAN_T4 #5: Messages from ECU are partially lost.

sepp89117, 2020-01-05:
I communicate with an engine control unit and some messages from the ECU do not arrive. My Teensy runs at 600 Mhz and CAN at 60 Mhz.
To retrieve received messages I use without a delay in a while loop:
`if (opcCan.read (can_MsgRx)) handleDataMsg(can_MsgRx);`

Initially I used Teensy to monitor communication with an ECU tester and the ECU. I already had data loss here until I set `opcCan.setMaxMB(1)`. Since Teensy is now the tester, it is no longer possible to use just one mailbox.
- What can I do so that I no longer miss any messages? 
- How many MBs are generated during CAN initiation? 
- Is there a way to get the messages in the order they are received? 

An example of how to work with the library in such a case would be great.

My setup is:
```cpp
FlexCAN_T4<CAN2, RX_SIZE_256, TX_SIZE_128> opcCan;
uint32_t canBaud = (uint32_t)500000;

void setup() {
  Serial.begin(9600);

  //init CAN
  opcCan.begin();
  opcCan.setBaudRate(canBaud);
  opcCan.setClock(CLK_60MHz);
}
```

I thank you in advance for the help

tonton81, 2020-01-05:
You could enable interrupts for mailboxes too or enable FIFO which stores messages in a 6 deep hardware queue. Usually polling is not ideal if frames are critical because all flagged mailboxes would have to be checked until a message is seen or not, it is better to have the interrupts enabled so the mailboxes could be processed immediately upon reception by the ISR. You could also limit the frames by using filters if you are looking for specific ones, that will be done in hardware.

tonton81, 2020-01-05:
`#include <FlexCAN_T4.h>
FlexCAN_T4<CAN2, RX_SIZE_256, TX_SIZE_16> Can0;

void setup(void) {
  Serial.begin(115200); delay(400);
  pinMode(6, OUTPUT); digitalWrite(6, LOW); // enable tranceiver
  Can0.begin();
  Can0.setBaudRate(500000);
  Can0.setMaxMB(16); // up to 64 max for T4, not important in FIFO mode, unless you want to use additional mailboxes with FIFO
  Can0.enableFIFO();
  Can0.enableFIFOInterrupt();
  Can0.onReceive(canSniff);
  Can0.mailboxStatus();
}

void canSniff(const CAN_message_t &msg) {
  Serial.print("MB "); Serial.print(msg.mb);
  Serial.print("  OVERRUN: "); Serial.print(msg.flags.overrun);
  Serial.print("  LEN: "); Serial.print(msg.len);
  Serial.print(" EXT: "); Serial.print(msg.flags.extended);
  Serial.print(" TS: "); Serial.print(msg.timestamp);
  Serial.print(" ID: "); Serial.print(msg.id, HEX);
  Serial.print(" Buffer: ");
  for ( uint8_t i = 0; i < msg.len; i++ ) {
    Serial.print(msg.buf[i], HEX); Serial.print(" ");
  } Serial.println();
}

void loop() {
  Can0.events();
}
`

sepp89117, 2020-01-05:
Thank you for the fast answer!
I adjusted my code and tested it briefly. So far it looks very good. I give feedback again after extensive tests.
If everything goes well, I will publish my code. It is an interesting project!

sepp89117, 2020-01-19:
Hello, I was able to test the code sufficiently. Everything works as intended. **Thanks a lot!**
Here is the link to my project:
https://github.com/sepp89117/Opel-Astra-H-opc-CAN-Gauge

closed 2020-05-17 by tonton81