tonton81/FlexCAN_T4 #6: Multiple Definition Compilation Error

msfrt, 2020-01-11:
I'm attempting to include this into another class definition outside of the main sketch file. There are no compilation errors when the T4 library is included in the main sketch, however, I get them when included in other header/implementation files:

```
/var/folders/zh/53jh0mfj1jlf_blr60ffjhqh0000gn/T/arduino_build_650974/sketch/StateCAN.ino.cpp.o: In function `flexcan_isr_can3fd()':
/Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4/FlexCAN_T4FD.tpp:403: multiple definition of `flexcan_isr_can3fd()'
/var/folders/zh/53jh0mfj1jlf_blr60ffjhqh0000gn/T/arduino_build_650974/sketch/StateCAN.cpp.o:/Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4/FlexCAN_T4FD.tpp:403: first defined here
/Applications/Arduino.app/Contents/Java/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Disabling relaxation: it will not work with multiple definitions
/var/folders/zh/53jh0mfj1jlf_blr60ffjhqh0000gn/T/arduino_build_650974/sketch/StateCAN.ino.cpp.o: In function `flexcan_isr_can3fd()':
/Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4/FlexCAN_T4FD.tpp:403: multiple definition of `_CAN3'
/var/folders/zh/53jh0mfj1jlf_blr60ffjhqh0000gn/T/arduino_build_650974/sketch/StateCAN.cpp.o:/Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4/FlexCAN_T4FD.tpp:403: first defined here
/var/folders/zh/53jh0mfj1jlf_blr60ffjhqh0000gn/T/arduino_build_650974/sketch/StateCAN.ino.cpp.o: In function `flexcan_isr_can3fd()':
/Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4/FlexCAN_T4FD.tpp:403: multiple definition of `_CAN2'
/var/folders/zh/53jh0mfj1jlf_blr60ffjhqh0000gn/T/arduino_build_650974/sketch/StateCAN.cpp.o:/Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4/FlexCAN_T4FD.tpp:403: first defined here
/var/folders/zh/53jh0mfj1jlf_blr60ffjhqh0000gn/T/arduino_build_650974/sketch/StateCAN.ino.cpp.o: In function `flexcan_isr_can3fd()':
/Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4/FlexCAN_T4FD.tpp:403: multiple definition of `_CAN1'
Multiple libraries were found for "FlexCAN_T4.h"
 Used: /Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4
/var/folders/zh/53jh0mfj1jlf_blr60ffjhqh0000gn/T/arduino_build_650974/sketch/StateCAN.cpp.o:/Users/daveyonkers/Documents/Arduino/libraries/FlexCAN_T4/FlexCAN_T4FD.tpp:403: first defined here
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 4.0.
```

tonton81, 2020-01-11:
I believe i followed this post already on stack exchange i think. Can you zip up your project and send it to me so i can check what your trying to include in your sketch if its done correctly?
Yodabox64@hotmail.com

msfrt, 2020-01-11:
I did not see the post on Stack Exchange, so I'll be sure to search for it. I'll send you the project now.

tonton81, 2020-01-12:
I declared the objects and background handlers as static, this compiles your code correctly now. I also tested it with a current CAN2.0 test code between my own project and it works well. Try out the new patch...

msfrt, 2020-01-13:
Great, thank you so much!

closed 2020-05-17 by tonton81