tonton81/FlexCAN_T4 #7: CAN1 not functional, but CAN2 and CAN3 are

mcguire-steve, 2020-01-13:
Hi there! CAN1 appears to not work for me. Adjusting the MWE appropriately, I can see output on the pins associated with CAN2 (1) and CAN3 (31), but nothing shows up under CAN1 (22). I'm using my scope just to verify that the Teensy hardware is outputting something. Any ideas? I just pulled from master, I'm at commit 017d255811c9a3a20ea81c23721d2526ef077a60. I've validated that I can wiggle the pins manually via GPIO and see a correct response, so the output drivers aren't burned out. Same thing with a new, fresh-out-of-the-bag T4. Any ideas?

```
#include <FlexCAN_T4.h>

FlexCAN_T4<CAN1, RX_SIZE_256, TX_SIZE_16> can;
uint8_t ledState = HIGH;

void setup() {

  Serial.begin(115200);
  can.begin();
  can.setBaudRate(1000000);
    
  pinMode(13, OUTPUT);
  digitalWrite(13, HIGH);
}

void loop() {
  digitalWrite(13, ledState);
  ledState = !ledState;

  can.mailboxStatus();
  //Send some can messages
  //Push a can msg
  CAN_message_t msg;
  msg.id = random(0x1,0x7FE);
  for ( uint8_t i = 0; i < 8; i++ ) msg.buf[i] = i + 1;
  can.write(msg);
  Serial.println("Pushed CAN msg");
  delay(100);  
}
```

mcguire-steve, 2020-01-13:
I should also point out that I've seen intermittent failures on the CAN2 and CAN3 interfaces - leaving the chip sit for a bit apparently allows something to reset and re-enable. Never for CAN1 tho.

tonton81, 2020-01-13:
That sounds really strange. I am using BCAN at 125kbps and CAN2 at 500kbps at the same time in an automotive environment

tonton81, 2020-01-13:
`#include <FlexCAN_T4.h>
FlexCAN_T4<CAN1, RX_SIZE_256, TX_SIZE_16> Can0;

void setup(void) {
  Serial.begin(115200); delay(400);
  pinMode(6, OUTPUT); digitalWrite(6, LOW); // enable tranceiver
  pinMode(13, OUTPUT); digitalWrite(13, LOW); // enable tranceiver
  Can0.begin();
  //  Can0.setBaudRate(500000);
  Can0.setClock(CLK_60MHz);

  Can0.setBaudRate(125000);
  Can0.setMaxMB(16); // up to 64 max for T4, not important in FIFO mode, unless you want to use additional mailboxes with FIFO
  Can0.enableFIFO();
  Can0.enableFIFOInterrupt();
//  Can0.enableMBInterrupts();
  Can0.onReceive(canSniff);
  Can0.mailboxStatus();
}

void canSniff(const CAN_message_t &msg) {
  Serial.print("MB "); Serial.print(msg.mb);
  Serial.print("  OVERRUN: "); Serial.print(msg.flags.overrun);
  Serial.print("  LEN: "); Serial.print(msg.len);
  Serial.print(" EXT: "); Serial.print(msg.flags.extended);
  Serial.print(" TS: "); Serial.print(msg.timestamp);
  Serial.print(" ID: "); Serial.print(msg.id, HEX);
  Serial.print(" Buffer: ");
  for ( uint8_t i = 0; i < msg.len; i++ ) {
    Serial.print(msg.buf[i], HEX); Serial.print(" ");
  } Serial.println();
}

void loop() {
  Can0.events();

  //  static uint32_t timeout = millis();
  //  if ( millis() - timeout > 800 ) { // send random frame every 20ms
  //    digitalWrite(13, !digitalRead(13));
  //    CAN_message_t msg;
  //    msg.id = random(0x1, 0x7FE);
  //    for ( uint8_t i = 0; i < 8; i++ ) msg.buf[i] = i + 1;
  //    Can0.write(msg);
  //    timeout = millis();
  //  }

}
`

klaasdc, 2020-05-01:
I have a similar problem, but on both CAN1 & CAN2, connected to ADM3053's. Eventually I found this: I put the T4 on the desk with only USB cable and scope leads to CAN1 TX. Nothing visible on TX even though sending in a loop. Once I briefly short TX1 to RX1, CAN traffic starts to appear on TX1. Pressing reset at this point keeps it going, but after re-plugging the USB it stays silent. Shorting RX/TX again makes CAN traffic appear.
Is there supposed to be additional pullup/pulldowns to the CAN TX lines?

tonton81, 2020-05-01:
Both endpoints should have resistors, its a differential network, no resistors, TX waits forever...

klaasdc, 2020-05-02:
> 
> 
> Both endpoints should have resistors, its a differential network, no resistors, TX waits forever...

Ah, I assumed the differential signalling was only generated inside the transceivers, from the TX/RX lines. Makes sense now.

tonton81, 2020-05-02:
Yeah the resistors have to go between TX and RX wires at each furthest node/device, no pullup or pulldowns. Any node inside the endpoints dont need resistors

mcguire-steve, 2020-11-10:
So it turns out I missed a critical feature of CAN, in that one must have at least two nodes on the network to receive the acknowledge signal. Everything works great. Appreciate the work on this library!

closed 2020-11-10 by mcguire-steve