tonton81/FlexCAN_T4 #8: Issus with filters and mailboxes

robinhellstrom, 2020-03-24:
I've having trouble getting CAN-messages through the mailboxes that I try to filter out messages with.

Maybe you could clarify in the **README.md** more about how the thoughts are on filters and mailboxes.

When I set up the code with this settings then will I only get msg 256 on CAN1 and nothing on CAN2 or CAN3:

````
#include <FlexCAN_T4.h>

// CAN creation
FlexCAN_T4<CAN1, RX_SIZE_256, TX_SIZE_16> CAN_1;
FlexCAN_T4<CAN2, RX_SIZE_256, TX_SIZE_16> CAN_2;
FlexCAN_T4<CAN3, RX_SIZE_256, TX_SIZE_16> CAN_3;

// CAN-messages (1&2)
static CAN_message_t TX_210;
static CAN_message_t TX_256;
static CAN_message_t TX_500;

// CAN1-messages
static CAN_message_t TX_310;
static CAN_message_t TX_704;
static CAN_message_t TX_705;
static CAN_message_t TX_260;
static CAN_message_t TX_261;

// CAN2-messages
static CAN_message_t TX_122; 

// CAN3-messages
static CAN_message_t TX_021204B8;
static CAN_message_t TX_03C3F7FC;
static CAN_message_t TX_09C050B8;
static CAN_message_t TX_0E03D7F8;
static CAN_message_t TX_0EE0500A;
static CAN_message_t TX_10500048; 
static CAN_message_t TX_13F041BA; 
static CAN_message_t TX_15B04028;
static CAN_message_t TX_16801028;
static CAN_message_t TX_1760E008;
static CAN_message_t TX_19500008;
static CAN_message_t TX_19F010FC;
static CAN_message_t TX_1093F7FC;
static CAN_message_t TX_1163F7FC;
static CAN_message_t TX_000FFFFE;

setup() {

    // Configure CAN interfaces
    CAN_1.begin();
    CAN_1.setBaudRate(500000);
    CAN_1.setMBFilter(REJECT_ALL);
    CAN_1.setMBFilterRange(MB0, 0x255, 0x710);
    CAN_1.setMBFilter(MB0, 0x255);
    CAN_1.setMBFilter(MB1, 0x258);
    CAN_1.setMBFilter(MB2, 0x259);
    CAN_1.setMBFilter(MB3, 0x26A);
    CAN_1.setMBFilter(MB4, 0x300);
    CAN_1.setMBFilter(MB5, 0x301);
    CAN_1.setMBFilter(MB6, 0x302);
    CAN_1.setMBFilter(MB7, 0x501);
    CAN_1.setMBFilter(MB8, 0x502);
    CAN_1.setMBFilter(MB9, 0x710);
    CAN_1.enhanceFilter(MB0);
    CAN_1.enhanceFilter(MB1);
    CAN_1.enhanceFilter(MB2);
    CAN_1.enhanceFilter(MB3);
    CAN_1.enhanceFilter(MB4);
    CAN_1.enhanceFilter(MB5);
    CAN_1.enhanceFilter(MB6);
    CAN_1.enhanceFilter(MB7);
    CAN_1.enhanceFilter(MB8);
    CAN_1.enhanceFilter(MB9);
    
    CAN_2.begin();
    CAN_2.setBaudRate(500000);
    CAN_2.setMBFilter(REJECT_ALL);
    CAN_2.setMBFilter(MB0, 0x250, STD);
    CAN_2.setMBFilter(MB1, 0x255, STD);
    CAN_2.setMBFilter(MB2, 0x258, STD);
    CAN_2.setMBFilter(MB3, 0x259, STD);
    CAN_2.setMBFilter(MB4, 0x501, STD);
    CAN_2.setMBFilter(MB5, 0x502, STD);
    CAN_2.enhanceFilter(MB0);
    CAN_2.enhanceFilter(MB1);
    CAN_2.enhanceFilter(MB2);
    CAN_2.enhanceFilter(MB3);
    CAN_2.enhanceFilter(MB4);
    CAN_2.enhanceFilter(MB5);
    
    CAN_3.begin();
    CAN_3.setBaudRate(125000);
    CAN_3.setMB(MB20, TX);
    CAN_3.setMB(MB21, TX);
    CAN_3.setMB(MB22, TX);
    CAN_3.setMB(MB23, TX);
    CAN_3.setMB(MB24, TX);
    CAN_3.setMB(MB25, TX);
    CAN_3.setMB(MB26, TX);
    CAN_3.setMB(MB27, TX);
    CAN_3.setMB(MB28, TX);
    CAN_3.setMB(MB29, TX);
    CAN_3.setMB(MB30, TX);
    CAN_3.setMB(MB31, TX);
    CAN_3.setMB(MB32, TX);
    CAN_3.setMB(MB33, TX);
    CAN_3.setMB(MB34, TX);
    CAN_3.setMB(MB35, TX);
    CAN_3.setMB(MB36, TX);
    CAN_3.setMBFilter(REJECT_ALL);
    CAN_3.setMBFilter(MB4, 0x0C505226, EXT);    // MB4-7 are extended frame
    CAN_3.setMBFilter(MB5, 0x1EF00066, EXT);
    CAN_3.setMBFilter(MB6, 0x1F200002, EXT);
    CAN_3.enhanceFilter(MB4);
    CAN_3.enhanceFilter(MB5);
    CAN_3.enhanceFilter(MB6);
}

void checkCAN() {

  CAN_message_t msg;

  if (CAN_1.read(msg)) {
    Serial.print("CAN 1 : ");
    Serial.println(msg.id, HEX);
    switch (msg.id) {
      case 0x258:
        //Do stuff...
        break;
      case 0x259:
        //Do stuff...
        break;
      case 0x255:
        //Do stuff...
        break;
      case 0x26A:
        //Do stuff...
        break;
      case 0x300:
        //Do stuff...
        break;
      case 0x301:
        //Do stuff...
        break;
      case 0x302:
        //Do stuff...
        break;
      case 0x501:
        //Do stuff...
        break;
      case 0x502:
        //Do stuff...
        break;
      case 0x710:
        //Do stuff...
        break;

      default:
        break;
    }
  }

  if (CAN_2.read(msg)) {
    Serial.print("CAN 2 : ");
    Serial.println(msg.id, HEX);
    switch (msg.id) {
      case 0x250:
        //Do stuff...
        break;
      case 0x258:
        //Do stuff...
        break;
      case 0x259:
        //Do stuff...
        break;
      case 0x255:
        //Do stuff...
        break;
      case 0x501:
        //Do stuff...
        break;
      case 0x502:
        //Do stuff...
        break;
      default:
        break;

    }
  }

  if (CAN_3.read(msg)) {
    Serial.print("CAN 3 : ");
    Serial.println(msg.id, HEX);
    switch (msg.id) {
      case 0x1EF00066:
        //Do stuff...
        break;
      default:
        break;

    }
  }
}

void loop() {
    checkCAN();
}
````
When I change the seup to this do I get all the messages:

````
setup() {

    // Configure CAN interfaces
    CAN_1.begin();
    CAN_1.setBaudRate(500000);
    CAN_1.setMBFilter(ACCEPT_ALL);
    .......
````

tonton81, 2020-03-24:
All messages from CAN1 you mean? Or all 3 busses?

robinhellstrom, 2020-03-24:
@tonton81 there was a typo, I'm not using FilterRange at the same time as regular filters. All CAN1 traffic except 0x258 disappears when using the filters above. 

Are there any special MB:s that don't work with standard frame messages?

tonton81, 2020-03-24:
Not that i am aware of. All busses use the same code, all controllers should respond the same. You can print out the mailbox info with mailboxStatus() to confirm mailboxes should be all set to standard and not extended, else the filtering depends on that flag

robinhellstrom, 2020-03-24:
I'll give your suggestions a go tomorrow, do I command the mailboxStatus() after I've initiated Serial.begin() in the setup()-loop?

tonton81, 2020-03-25:
After yes

closed 2021-01-03 by tonton81