tonton81/FlexCAN_T4 #12: Teensy 3.6 - Both CAN channels not working

juleq, 2020-05-23:
Hi there,

I am trying your library with a Teensy 3.6. I use Platform.IO. I have had no luck getting frames out of my transcievers yet (Dual CAN transcievers from Tindie, CAN0 uses ALT).

Are there any steps that I need to do when I want to use it on 3.6? I saw the *.zip file and also tried that. It did behave differently in that it offered setRx and setTx, so I did that for CAN0. But no matter what, I could not get a single frame, be it rx or tx.

I have tried CAN0 and CAN1, no luck transmitting on either of them. I have other nodes on the bus and a PCAN interface. The bus does not show any errors. Baud rate is correct, termination is correct. 

I started off using the FIFO example. I call the events method about every 100ms. I assume that the onReceive callback is decoupled from the interrupt?

Do you have an idea what I might have missed? Do I have to set the clock for the CAN peripheral or something? Anything with respect to pin mapping?

Thanks for helping out. BR

tonton81, 2020-05-23:
my friend Joel has it running on a T3.2, make sure you are using the FlexCAN_T4 constructor, not the FlexCAN_T4FD constructor

juleq, 2020-05-23:
Ok, will double check our suggestions. I think the transcievers do just have an Rs to select high speed/low power, not an En. I have not selected one yet, will go for the high speed. Can I use the version in the project root or is it still required to unpack the zip for 3.x?

juleq, 2020-05-23:
Yap, seems to have been my neglegence of the Rs pin. It is transmitting now. Thanks. When I understand your library correctly, all of the API is decoupled from interrupt? By calling events() and getting callbacks for all queued frames? Or is there a way to omit calling events() and just get called in interrupt context?

tonton81, 2020-05-23:
yes that is correct. the interrupt queues, events dequeues. i will work on implementing direct interrupt handling without queues eventually, for now you can use loop or a timer to run events

juleq, 2020-05-23:
Ok. Had the idea just using TimerInterval and decoupling from timer ISR myself. So that might be a viable option?

But maybe adapting my main loop to your strategy might be a better fit. I think its probably good practice to keep the interrupt context private like you did. I am using the interrupt on an Arduino to ensure that I miss no frames. I need to integrate a bunch of 10ms signals.

I assume the teensy is plenty fast to ensure the rx buffer is always as good as empty. 

One other question: Events will dequeue all pending events in order, right? So one call to events will get me multiple callbacks. And the order is FIFO when is use that?

tonton81, 2020-05-23:
yes you can direct from isr, however, we want isr to be as fast as possible. if a user has delays in his callback code, the isr will not release until it's complete. events dequeues one frame per refresh. you can poll events till theyre empty, events() returns the size of the rx queue bits 12-23, bits 0 to 11 show how many transmits are left in the queue

so if you wanted to clear the whole rx queue in one events(), you could mask bits 12 to 23 of the return
while ( events() & 0xFFF000 ); // this clears all receptions in the queue

juleq, 2020-05-23:
Ok, reworked my poor-mans scheduling strategy to make better use of your API. Works solid. Great. Thanks for that.

closed 2020-05-23 by juleq