tonton81/FlexCAN_T4 #14: ALT pins on Teensy 3.5 and 3.6 are not set to GPIO state.

opelpanfan, 2020-07-07:
ALT pins not available as GIO pins after CAN started. 
Pin 29 and 30 affected on Teensy 3.5 and 3.6

PR ALT_35_36 raised with a fix in
FlexCAN_T4.tpp lines 576, 579, 644, 647 I believe are wrong as pins set to 0
pins should be reset back to GPIO state using PORT_PCR_MUX(1)

tonton81, 2020-07-07:
default state for gpios is DISABLED, why convert them to GPIO when they (all pins) are disabled by default?

opelpanfan, 2020-07-08:
Because if you declared pins in setup before you do CAN init they will be
put back to disable state.

tonton81, 2020-07-08:
thanks for the info. i would suggest a different method then that won't affect default state (or unset). i will check the RM later to see what conditions we can check first, to see whether the pin was used or was not used by flexcan, or if it was set other than default.

tonton81, 2020-07-08:
I just did a commit for the pin configuration validation for T3. Can you let me know what you think?

If the default pin is NOT set, begin uses DEF as default.
If the default pin was set to GPIO mode, begin() will call ALT automatically.
If default and alternate pins are both used, ALT is automatically set and override ALT pin state to take control.
using setTX and setRX manually after that is fine and your selections of ALT or DEF are enforced. If the previous pin was forced to flexcan mode, it will be set to 0.

opelpanfan, 2020-07-08:
Sounds good. Tested on my board and it works thank you. Closed PR as you did a better job handling this issue :)

tonton81, 2020-07-08:
Excellent. I'll have to add the T4.x ones later on, it's 4AM :P

opelpanfan, 2020-07-08:
All sorted for T3.5 and 3.6

closed 2020-07-08 by opelpanfan