tonton81/FlexCAN_T4 #16: Example sketch crashes

larsrengersen, 2020-09-04:
I have a little test setup on the bench.
Node 1 = CAN sender, has 120 Ohm termination
Node 2 = Teensy 3.6 with ISO1050 transceiver
Node 3 = CAN sniffer, also has 120 Ohm termination

I can see the traffic Node 1 sends on Node 3, but on the Teensy nothing.
Even worse, when I just uploaded the example CAN2.0_example_FIFO_with_interrupts.ino my Macbook crashed and gave me a high fan for a second and then a black screen (high USB current??)
Crash report starts with
`panic(cpu 1 caller 0xffffff80062469aa): Kernel trap at 0xffffff80066ab9a3, type 14=page fault, registers:`
Earlier I just got nothing back on the serial monitor and after a short while the Teensyduino App crashed. Upon restart I got:
`Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
    at cc.arduino.view.NotificationPopup.updateLocation(NotificationPopup.java:244)
    at cc.arduino.view.NotificationPopup.<init>(NotificationPopup.java:222)
    at cc.arduino.view.NotificationPopup.<init>(NotificationPopup.java:72)
    at cc.arduino.contributions.ContributionsSelfCheck.lambda$run$0(ContributionsSelfCheck.java:134)
    at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
    at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
    at java.awt.EventQueue.access$500(EventQueue.java:97)
    at java.awt.EventQueue$3.run(EventQueue.java:709)
    at java.awt.EventQueue$3.run(EventQueue.java:703)
    at java.security.AccessController.doPrivileged(Native Method)
    at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
    at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
    at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
    at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
    at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
    at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
    at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
    at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)`
I also tried two IFCT examples, they too crash my mac.
I am using Teensyduino 1.51 and Arduino IDE 1.8.12 on a Macbook Pro using Catalina 10.15.6
Any thoughs on what is happening or how I can resolve these issues?

tonton81, 2020-09-04:
this is not related to the library, your mac is having java issues in general with arduino, try to update your java

larsrengersen, 2020-09-05:
Thanks for your quick response!
Since some (small) sketches worked and IFCT related did not I jumped to the conclusion it was the library.
But you are completely right, it was not.
I opened a topic at the PJRC forum and it turned out IDE 1.8.12 was causing the issues.

> IDE 1.8.12 was quickly replaced for perhaps related "issues" . Update the IDE to 1.8.13. Also TeensyDuino at ver 1.53
> 
> That looks like the mac OS version that comes with the integrated installer - so one stop at PJRC.com for IDE 1.53 it seems delivers complete IDE&TD install.
And indeed, did that upgrade and works like a charm now!
Thanks for your support and the great library.

closed 2020-09-05 by larsrengersen