tonton81/FlexCAN_T4 #19: problem reading msg identifier

leonardoBolzonella, 2021-02-02:
Hello, I have been testing FlexCAN using a Teensy3.6 to send data to a custom ECU that uses a custom can code.  I have changed the header file so that the msg.id is of type uint16_t, in order to match the custom can code. When a message is sent from the ECU with identifier 0x901 (1001 0000 0001), the teensy is reading id = 0x101  (001 0000 0001). I tried reverting to its original type (uint32_t) but nothing changed. Any idea why only 11 bits are being read in the id, instead of the 12 I was expecting? 

I hope the question is clear, 
Cheers,
Leo

tonton81, 2021-02-02:
try setting msg.flags.extended to 1

leonardoBolzonella, 2021-02-02:
Thanks for the quick reply! Should it set it in the frame being sent or is it a setting for the CAN receiver code? 
Sorry if this sounds like an obvious question but this is my first time using CAN protocol.

tonton81, 2021-02-02:
sending

leonardoBolzonella, 2021-02-02:
Worked, thank you!

closed 2021-02-02 by leonardoBolzonella