tonton81/FlexCAN_T4 #22: FlexCan for Teensy 4.1

ldevillez, 2021-04-04:
Hello do you know it the library works for a Teensy 4.1 ?

I used the demo code:

```
#include <FlexCAN_T4.h>
FlexCAN_T4<CAN3, RX_SIZE_256, TX_SIZE_16> Can0;

void setup(void) {
  Serial.begin(115200); delay(400);
  pinMode(6, OUTPUT); digitalWrite(6, LOW); /* optional tranceiver enable pin */
  Can0.begin();
  Can0.setBaudRate(50000);
  Can0.setMaxMB(16);
  Can0.enableFIFO();
  Can0.enableFIFOInterrupt();
  Can0.onReceive(canSniff);
  Can0.mailboxStatus();
}

void canSniff(const CAN_message_t &msg) {
  Serial.print("MB "); Serial.print(msg.mb);
  Serial.print("  OVERRUN: "); Serial.print(msg.flags.overrun);
  Serial.print("  LEN: "); Serial.print(msg.len);
  Serial.print(" EXT: "); Serial.print(msg.flags.extended);
  Serial.print(" TS: "); Serial.print(msg.timestamp);
  Serial.print(" ID: "); Serial.print(msg.id, HEX);
  Serial.print(" Buffer: ");
  for ( uint8_t i = 0; i < msg.len; i++ ) {
    Serial.print(msg.buf[i], HEX); Serial.print(" ");
  } Serial.println();
}

void loop() {
  Can0.events();

}
```

The preview of my can:
[image]

The blue output is connected to pin 30 (CRX3). I checked to have the same frequency on my can network.

Thanks in advance :)

tonton81, 2021-04-04:
yes 4.0 & 4.1 are working

ldevillez, 2021-04-04:
Okay thanks for your answer :)

ldevillez, 2021-04-04:
> yes 4.0 & 4.1 are working, what problem are you having? i see you have a baudrate of 50kbps, is that correct or typo for 500kbps?

I'm trying to use my computer to send a can packet (mainly standframe). For that i use a cheap usb-can translator (https://github.com/alexmohr/usb-can) then with can-utils i send a packet (red line on my screen is the CAN_L).

Then i used a SN65HVD230 as a can tansceiver. The Rx pin of this IC is then connected to the CRX of the T4.1 (I've tried the 3 controllers and I tried with another teensy).

So for now, I just try to print the receive can packet from the teensy. 

I used a baudrate of 50 kbps because it was easier for my scope (i configured my adaptor to works at this baudrate). I just tested with a baudrate of 500 kbps and i got also no result.

The test message that i use: "123#AABBCCDD"

msadie, 2021-04-04:
Can you confirm you have the CTX pin of teensy connected to Tx of
transceiver? This is required for receiving node to ack the message.
Also, be sure the Rs pin of transceiver is strongly pulled to ground.

ldevillez, 2021-04-04:
Thank you !! It was a bad solder joint of the Tx pin of the transceiver. I should have checked my joint :)

closed 2021-04-04 by ldevillez

Dreamcouple, 2022-08-16:
hii,
i am using teensy 4.1 with cubemars AK10-9 model motor , i just connect my motor high and low is connect to the waveshare can bus module , i try to read the data from motor can bus module to  teensy 4.1 but i am bot receiving any thing from teensy ?

ldevillez, 2022-08-16:
Hello @Dreamcouple,

Maybe create a new issue for your problem (i mean those 2 issues don’t seem to be related).
Also check your connexion, and the frequency of the system. It could be great to provide a schematic of your system :)

Dreamcouple, 2022-08-16:
[image]
here i upload my shcematic and flexcan t4 library program is uploaded into my teensy..

ldevillez, 2022-08-16:
Don’t use the Vin pin as a 3.3 source. Also did you check the number of 120 resistors ? Can you post (a simplified version) your code ?

Dreamcouple, 2022-08-16:
that can module have single 120ohm resistor , and i am not using vin pin for 3.3 v, i used 3.3v next to ground pin for supply of can bus module .. here i attached my code..
[image]

ldevillez, 2022-08-16:
If there is already one resistor in the Rlink and one in the motor, that makes one too much...
Also did you the frequency of the can bus ? Are you able to control the motor with the Rlink ? (and thus making sure that packets are send over the bus ?)

Dreamcouple, 2022-08-16:
Are you able to control the motor with the Rlink ?
i try to control the speed of motor , but i stuck in that problem , so you told already termination resistor in motor and rlink side adding of one more resistor in can bus module make problem this the thing you try to say??
Also did you the frequency of the can bus ? 
sry!! i am not get your point...

ldevillez, 2022-08-16:
Yes having more resistors can cause a problem. 
All the nodes in your can bus must have the same frequency (I saw your teensy running at 250 kHz).
But if you are not sure that messages are send to the bus, maybe investigate that first (maybe with an oscilloscope).

Dreamcouple, 2022-08-16:
[image]
i just removed 120ohm resistor of  my can bus module because two side of presenting termination resistor and also i changed the baudrate of my controller, but i'am not getting any value of output, i don't know why ,could you please helpme
[image]
,

ldevillez, 2022-08-16:
As I said, are you sure that trames are send over your bus ? Check with an oscilloscope.

Dreamcouple, 2022-08-16:
sryy, what is trames ?? or  did you ask about whether the motor controller sending the data or not ??

ldevillez, 2022-08-16:
Yes from what I understood the motor controller must send data over the can bus, is it the case ?

ldevillez, 2022-08-17:
Yes I understand that. Are frames send over the can bus in your setup (can you interact with the motor through your controller | so making your controller send can frames on the bus and the motor reading those frames) ?

Dreamcouple, 2022-08-17:
yes motor reading the frames and motor is running based on what parameters is set, but thing i'm not able to see the can bus frames read by motor or controllers..

ldevillez, 2022-08-17:
Ok great. Can you check now the Can Rx and Can Tx of ([SN65HVD230 CAN Board]) to see the waveforms ?

Dreamcouple, 2022-08-17:
nothingg, sir

Dreamcouple, 2022-08-17:
now i try to cmmunicate btw with two of my teensy 4.1 also not work, i dont what problem

ldevillez, 2022-08-17:
If there is no waveforms at CRx and CTx maybe your ([SN65HVD230 CAN Board]) is (part of) the problem.