tonton81/FlexCAN_T4 #23: CAN maximum baudrate

jimmie11, 2021-05-07:
This is my first ever project using the CAN bus. I am interfacing a CAN sensor to a Teensy 4.1 and the FlexCAN_T4 library.

My sensor can be configured with CAN baudrates of up to 1000 kbps. In my application, I need to use the fastest possible CAN baudrate.

My program works fine up to a CAN baudrate of 250 kbps which is unfortunately too slow in my case. Once I bump the sensor CAN rate beyond 250 kbps (options are 500, 1000), the Teensy4.1 stops receiving data.

Is there any way to bump the CAN baudrate on the Teensy4.1? Is this a hardware or a software limitation?

BTW, I am using the following board:

           http://copperhilltech.com/teensy-4-1-triple-can-bus-board-with-240x240-lcd-and-ethernet/

tonton81, 2021-05-07:
1000 does work, maybe the clock timings are different for your application, try changing the clock? can.setClock()

jimmie11, 2021-05-07:
Thank you.

I configured the sensor for 500 kbps then used the following code

```
  can1.begin();
  can1.setClock(60);
  can1.setBaudRate(500000);   //250000
```

designer2k2, 2021-07-11:
Running a teensy 4,0 here with the 24MHz default clock.  Read / send with 500kbps is no problem for me.

jimmie11, 2021-07-11:
Thank you @designer2k2. Yes, I am also confirming it.

It depends on both the sensor AND the hardware on the MCU side.  Basically the existence a 60 Ohm resistor across the CAN lines (on both sides, sensor and MCU).

tonton81, 2021-07-11:
you can get away with resistor less below 125kbps, but any higher requires it mandatory to function

closed 2023-04-14 by tonton81