tonton81/FlexCAN_T4 #24: Feature request, expose ESR and ECR register to diagnose software or hardware issues

designer2k2, 2021-07-09:
Hello,

to help with diagnosing issues (Software or Hardware related) the ECR and ESR register gives some insight. The ECR register holds the RX / TX Error Counters. And the ESR register holds a load of informations on the status of the CAN module. 
See Section "44.9.7 Error Counter Register (FLEXCANx_ECR)" and "44.9.8Error and Status 1 Register (FLEXCANx_ESR1)" in the documentation: https://www.pjrc.com/teensy/IMXRT1060RM_rev2.pdf
I would like to read this informations. And i believe this would also help others to see whats going on.

There is another thread in the PJRC forum mentioning this too: https://forum.pjrc.com/threads/24720-Teensy-3-1-and-CAN-Bus?p=103806&viewfull=1#post103806

There is some Code shown, what i hacked into my local FlexCAN_T4 lib, but im not shure if this is the appropiate way so i did not start a pull request:

Add to FlexCAN_T4.tpp:
````
FCTP_FUNC uint32_t FCTP_OPT::ErrorReport() {
	return (FLEXCANb_ESR1(_bus));
}

FCTP_FUNC uint8_t FCTP_OPT::ErrorCounter(bool RXorTX) {
  if (RXorTX == false) {
    return (FLEXCANb_ECR(_bus) >> 8); //shift 8 for RX
  }
  else {
    return (FLEXCANb_ECR(_bus));
  }
}
`````

Add to FlexCAN_T4.h: (Line 459, but not shure if correct here)
````
	uint32_t ErrorReport();
	uint8_t ErrorCounter(bool RXorTX);
````

With this an output like this is possible:
````
CAN RX Error Count: 0
CAN TX Error Count: 57
Error Status: 100000000110110
````

Looking forward on your feedback!

Anyway, thank you for this lib, i use it happily with a teensy 4 in my car and it works perfectly.

greetings

tonton81, 2021-07-09:
I would refrain from using those registers directly as at least one of them if i recall changes state when read. The ISR and the loop may cause rare inconsistancies so I would probably store them in a variable set to be read later, or have an error interrupt fire if necessary

designer2k2, 2021-07-10:
You are correct, i re-read the manual and reading the ESR register clears bits 15-10, clearing all the errors. Thats not helpful.
[image]

Reading the ECR for the RX/TX error counters is ok. They alone would help with finding troubles. So maybe this can considered?

Overall a error interrupt sounds good, or some other way where i can check if errors happened.

tonton81, 2021-07-11:
Would storing it be relevant? Or just for print debugging?
I'm testing something like this:

`FlexCAN State: Transmitting
FLT_CONF: Error Active - Indicates the Confinement State of the FLEXCAN module
Error Counter Registers:
 --> RX: 0
 --> TX: 0
`

designer2k2, 2021-07-11:
No good programmer here, so please take that as a wish on how it could be.

I am polling and currently only check the overrun flag.  
Would like to have something like the overrun flag what i could check if an error exists.

So im currently doing:
```
if (can1.read(canMsg)) {
  (fancy things)
}
```

and then could do something like checking a general :
```
if (can1.error) {
  (read deeper into the detailed errors and highlight to user)
}
```

Typically there is no access to the serial log in my usecase, its a automotive display, but it has a SD card. 
I show the "CAN Status", currently it just shows if CAN traffic exists, or if the overrun happens. 
That would then show a general "CAN Error" and i would dump more info to the logfile on the SD card.

Rate limiting the error messages should be my issue, i would possibly stop after a couple of seconds and ask the user to take action within my code.

By default this error reporting should be off, or that it must be actively checked.

Your example with clear text printing to the serial port would be a good example and starting point for users of this library!

tonton81, 2021-07-11:
You might get overruns depending on how busy your loop code is since you are polling frames. canMsg.overrun should return 1 or 0 if an overrun occured during that message reception, the errors are different than overrun

tonton81, 2021-07-12:
Okay I developed an error queue system based on changes of the ESR1. The ISR will push to the queue when an error is likely to occur, and your loop can pull it out, read the data from a struct, and optionally, internally print it to serial monitor with a bool overload

tonton81, 2021-07-12:
I've added the commits for error reporting, try it out :)

designer2k2, 2021-07-14:
this works great!

Testing it on a teensy 4  and unplugged the bus while running:
````
FlexCAN State: Receiving, BIT1_ERR, FRM_ERR, FLT_CONF: Error Active
FlexCAN State: Receiving, STF_ERR, FLT_CONF: Error Active
FlexCAN State: Receiving, BIT1_ERR, FRM_ERR, FLT_CONF: Error Active
FlexCAN State: Receiving, CRC_ERR, STF_ERR, FLT_CONF: Error Active
FlexCAN State: Receiving, BIT1_ERR, FRM_ERR, FLT_CONF: Error Active
FlexCAN State: Receiving, BIT1_ERR, FRM_ERR, RX_WRN: 97, FLT_CONF: Error Active
FlexCAN State: Receiving, BIT1_ERR, FRM_ERR, RX_WRN: 114, FLT_CONF: Error Active
FlexCAN State: Receiving, CRC_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, STF_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, FRM_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, STF_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, STF_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, FRM_ERR, RX_WRN: 131, FLT_CONF: Bus off
FlexCAN State: Receiving, CRC_ERR, RX_WRN: 131, FLT_CONF: Bus off
````

Thanks a lot for the really quick and great solution @tonton81  👍

bvernham, 2021-07-15:
If you go Buss off is there a way to reset the buss to see if this is an intermittent vs hardware issue?  In other words a retry?

For example I can have an OBD buss that I am monitoring then some one plugs a cheap code reader in which take the buss down.  When this is unplugged the buss can restart.  I like to joke if you want to cause a lot of chaos then just take a paper clip and short out the OBD buss then you will find how robust the whole vehicle system is.  For example the OBD bus is only dedicated to OBD or there is other control/monitoring traffic which is running on the buss also which will cause loss of communication between those modules.  Thoughts?  Thanks  Bruce

tonton81, 2021-07-15:
you can surely short the bus, it wont break anything, the controller hardware does the bus-off recovery by iso standards when valid CAN frame bits are seen on the bus

designer2k2, 2021-07-16:
@bvernham shorting out the CAN OBD on a relatively modern car (>2010) will only stop the OBD to work.
But if you would short out the Engine CAN on a modern VW/Audi then only a fault code reset will remove the dash lights, but the car works once the short is cleared.  (guess from where i know...) 

As @tonton81 said, the bus itself recovers with no need to actively reset/help.  

For me this feature request here can be closed.

bvernham, 2021-07-16:
Ahh, but the assumption here is the OEM hid all the control busses and only expose the OBD buss to potentially poor external CAN devices.

I can tell you this is not a fact for all OEM's.  Some have both OBD and inter-module CAN traffic on the same busses.  Which to me is a no no but it is the fact.  If you have 2 controllers that can not communicate with each other then for example an engine may go into a limp home mode with limited power/shifting and so on.

I am not saying this is the correct way to mechanize a vehicle CAN architecture but unfortunately some OEM's do.

If this T4.1 CAN controller will recover after the cause of the buss off issue is resolved then it is all good.

Thanks

Bruce

tonton81, 2021-07-16:
yes it auto recovers, it won't be iso standard if it wasn't. flexcan could, although it breaks standards, not exit from bus off, but requires a user to manually assert it to recover. this is not part of the library as it doesn't follow CAN device standards, and causes more code bloat and user involvement 😁

designer2k2, 2021-07-23:
As the request has been closed i close this Issue, thanks again @tonton81 👍

closed 2021-07-23 by designer2k2