tonton81/FlexCAN_T4 #27: Crashing when isotp.h is included in multiple project files

fredmcc, 2021-10-19:
Not sure if others have had this issue but when I include isotp.h in other project files (to use ISOTP_data variables in function calls) the Teensy4.1 is crashing when it receives a CAN frame. Just including the header file does this.
I haven't been able to work out why but it looks like static _ISOTP_OBJ is not being initialized. 
Could this be because the header file is included in a file that is not initializing the class. 
ie    isotp<RX_BANKS_16, 512> ISOtp;  ????

I might be doing something wrong but thought I would reach out. I'll try to look into it further.

Crash details below

=====================================================
addr2line to gives this.

Arduino\libraries\FlexCAN_T4/isotp.tpp:146

145: void ext_output2(const CAN_message_t &msg) {
146:   _ISOTP_OBJ->_process_frame_data(msg);
147: }

=====================================================
CrashReport:
  A problem occurred at (system time) 12:31:48
  Code was executing from address 0xC86
  CFSR: 82
	(DACCVIOL) Data Access Violation
	(MMARVALID) Accessed Address: 0x0 (nullptr)
	  Check code at 0xC86 - very likely a bug!
	  Run "addr2line -e mysketch.ino.elf 0xC86" for filename & line number.
  Temperature inside the chip was 50.57 °C
  Startup CPU clock speed is 600MHz
  Reboot was caused by auto reboot after fault or bad interrupt detected

tonton81, 2021-10-19:
have you tried the demo first to see if that works?

tonton81, 2021-11-21:
the object is being processed from a file or sketch? if from a file, the other files that might use it need to be defined as extern so if they use it it doesn't crash if it's a linking issue. the constructor must be defined somewhere but the object if used in other files (if not sketch) must be defined as extern so they can use it

DeanIsMe, 2022-07-19:
I've spent a long time debugging and found that it's a similar issue. 
My Teensy4.0 would crash when:
-  I compiled the program with FlexCAN_T4 included in more than 1 compilation unit, AND
- I sent a CAN packet to an empty CAN bus. It's not ACKed, resulting in infinitely repeating CAN message. (The repeating CAN message isn't the problem - the program crash is).

Note that the issue only occurred if FlexCan_T4 code was _included_ in multiple compilation units - it didn't matter whether or not the code was actually _executed_.

I'm guessing there are some issues regarding static variables and functions in this lib.

My solution was to only `#include <FlexCAN_T4.h>` in a SINGLE compilation unit (`.cpp` or `.ino` file).

tonton81, 2022-07-19:
well usually yes they should only be in one compilation unit.
it is possible that it's because it's a template class, functions and variables are created for that compile time object specifically. now the other unit that attempted to load that, is not awaye of the compile time object because it was created elsewhere, so those functions and variables are not of that object's "type"

so while one compilation unit knows the object which is created at compile time, the other unit is not aware of it's type or compiled functions etc.. probably these 2 units are clashing, one accepting the type, and the other just choking. A runtime object doesn't see this issue because the class has fixed functions for all objects so all units compiled the necessary functions for that object to use.

how are you creating your object currently? are you assigning a base pointer? you should only create ONE template instance per bus in ONE compilation unit. having 2 different units create one for same bus will cause issues that is for sure. creating a base class pointer should allow you to create an extern base pointer in your other units and point it to the template's object. you can also use the background preset pointers (_CAN1,_CAN2,_CAN3), take note of the "_". the moment you create a template object for a bus, you can use the pre-assigbed base pointer (or make one of your own). the library will assign your template object automatically to the _CAN* pointers.

closed 2023-04-14 by tonton81

dylanmeehan, 2023-09-21:
Is the problem here that isotp.tpp defines ext_output2(), which is then included in isotp.h . So, if multiple files include isotp.h, then there are multiple definitions of ext_output2()? 

Here is the definition in isotp.tpp. isotp.h includes isotp.tpp
```
void ext_output2(const CAN_message_t &msg) {
  _ISOTP_OBJ->_process_frame_data(msg);
}
```
In my project, including isotp.h in multiple other .h files causes the FlexCAN_T4 library to crash when a can callback is triggered. Interestingly, including isotp.h in multiple  .h files doesn't cause any linker error. Maybe this is because FlexCAN_T4.h  declares ext_output2 as extern void in FlexCan_T4.h? 

I think I fix this issue by declaring ext_output2 as inline in isotp.tpp . At least, such a fix prevents the library from crashing on can callbacks. But, I have not tested the isotp functionality. 

That is, at the end of isotp.tpp I added inline to get: 
`inline void ext_output2(const CAN_message_t &msg) {
  _ISOTP_OBJ->_process_frame_data(msg);
}
`
Is the ext_output2 even used by FlexCAN_T4? I see it is called in FlexCAN_T4.tpp . But, the readme says that these weak functions are intended to be used by other libraries. So, is ext_output2 even needed? Is it only used for debugging? Should it simply be removed from isotp.tpp?

If this fix seems good, I can generate a pull request to add the inline keyword or (if appropriate) remove the ext_output2 definition from isotp.tpp entirely.