tonton81/FlexCAN_T4 #29: CANFD data length not OK if longer then 8 Byte

GitSchorschGit, 2021-11-12:
Hi All,
I tested the CANFD and it works very well, but when the Data is longer then 8 Bytes
maybe 24, then on the bus are only 8 Bytes.
For example:
msg.len = 24;    -> CAN-Bus shows only 8 Bytes

Thanks for the good work!

chaosu-zd, 2021-11-12:
[image]

GitSchorschGit, 2021-11-12:
Thank you for your answer! I tried it and it works very well!
How can I find this documentation and how can I calculate the example above?
Is 896 Bytes the max value? Then I have to divide it by 8, 16, 32 or 64. Is that right?
Thanks for all!

tonton81, 2021-11-12:
The mailbox has to support the size you are using, by default each mailbox is 8 bytes for 64 mailboxes. The more bytes you need, the more mailboxes it consumes. So at 64 byte mailboxes among 2 memory regions, that leaves you a total of 14 mailboxes. The amount of mailboxes available for a specific size is returned by the same function setRegions(). It should be in the readme on this repo

closed 2021-11-21 by tonton81