tonton81/FlexCAN_T4 #33: no matching function for call to 'FlexCAN_T4<(CAN_DEV_TABLE)1075642368u, (FLEXCAN_RXQUEUE_TABLE)256u, (FLEXCAN_TXQUEUE_TABLE)16u>::setMBFilterProcessing(FLEXCAN_MAILBOX, int)'

Manivannan-maker, 2021-12-17:
In mailbox_filtering_example_with_interrupts example , for Teensy4.1 board

The error occuring at this line "  Can0.setMBFilterProcessing(MB0,0x00);"
The complete error message is listed below.
But when I use Can0.setMBFilter(MB0,0x00); instead of Can0.setMBFilterProcessing(MB0,0x00); " there is no compilation error . 
Could you explain the difference between setMBFilter and setMBFilterProcessing . also the cause of this compilation error.

Arduino: 1.8.16 (Windows 10), TD: 1.55, Board: "Teensy 4.1, Keyboard + Mouse + Touch Screen, 600 MHz, Faster, US English"

mailbox_filtering_example_with_interrupts: In function 'void setup()':

C:\Users\MANIVA~1\AppData\Local\Temp\arduino_modified_sketch_668396\mailbox_filtering_example_with_interrupts.ino:12:24: warning: invalid conversion from 'int' to 'FLEXCAN_MAILBOX' [-fpermissive]

     Can0.setMB(i,RX,EXT);

                        ^

In file included from C:\Users\MANIVA~1\AppData\Local\Temp\arduino_modified_sketch_668396\mailbox_filtering_example_with_interrupts.ino:1:0:

C:\Users\manivannans\sketchbook\libraries\FlexCAN_T4-master/FlexCAN_T4.h:302:18: note:   initializing argument 1 of 'bool FlexCAN_T4<_bus, _rxSize, _txSize>::setMB(const FLEXCAN_MAILBOX&, const FLEXCAN_RXTX&, const FLEXCAN_IDE&) [with CAN_DEV_TABLE _bus = (CAN_DEV_TABLE)1075642368u; FLEXCAN_RXQUEUE_TABLE _rxSize = (FLEXCAN_RXQUEUE_TABLE)256u; FLEXCAN_TXQUEUE_TABLE _txSize = (FLEXCAN_TXQUEUE_TABLE)16u; FLEXCAN_MAILBOX = FLEXCAN_MAILBOX; FLEXCAN_RXTX = FLEXCAN_RXTX; FLEXCAN_IDE = FLEXCAN_IDE]'

 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>

                  ^

C:\Users\manivannans\sketchbook\libraries\FlexCAN_T4-master/FlexCAN_T4.tpp:242:16: note: in expansion of macro 'FCTP_OPT'

 FCTP_FUNC bool FCTP_OPT::setMB(const FLEXCAN_MAILBOX &mb_num, const FLEXCAN_RXTX &mb_rx_tx, const FLEXCAN_IDE &ide) {

                ^

C:\Users\MANIVA~1\AppData\Local\Temp\arduino_modified_sketch_668396\mailbox_filtering_example_with_interrupts.ino:15:24: warning: invalid conversion from 'int' to 'FLEXCAN_MAILBOX' [-fpermissive]

     Can0.setMB(i,TX,EXT);

                        ^

In file included from C:\Users\MANIVA~1\AppData\Local\Temp\arduino_modified_sketch_668396\mailbox_filtering_example_with_interrupts.ino:1:0:

C:\Users\manivannans\sketchbook\libraries\FlexCAN_T4-master/FlexCAN_T4.h:302:18: note:   initializing argument 1 of 'bool FlexCAN_T4<_bus, _rxSize, _txSize>::setMB(const FLEXCAN_MAILBOX&, const FLEXCAN_RXTX&, const FLEXCAN_IDE&) [with CAN_DEV_TABLE _bus = (CAN_DEV_TABLE)1075642368u; FLEXCAN_RXQUEUE_TABLE _rxSize = (FLEXCAN_RXQUEUE_TABLE)256u; FLEXCAN_TXQUEUE_TABLE _txSize = (FLEXCAN_TXQUEUE_TABLE)16u; FLEXCAN_MAILBOX = FLEXCAN_MAILBOX; FLEXCAN_RXTX = FLEXCAN_RXTX; FLEXCAN_IDE = FLEXCAN_IDE]'

 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>

                  ^

C:\Users\manivannans\sketchbook\libraries\FlexCAN_T4-master/FlexCAN_T4.tpp:242:16: note: in expansion of macro 'FCTP_OPT'

 FCTP_FUNC bool FCTP_OPT::setMB(const FLEXCAN_MAILBOX &mb_num, const FLEXCAN_RXTX &mb_rx_tx, const FLEXCAN_IDE &ide) {

                ^

mailbox_filtering_example_with_interrupts:22: error: no matching function for call to 'FlexCAN_T4<(CAN_DEV_TABLE)1075642368u, (FLEXCAN_RXQUEUE_TABLE)256u, (FLEXCAN_TXQUEUE_TABLE)16u>::setMBFilterProcessing(FLEXCAN_MAILBOX, int)'

  Can0.setMBFilterProcessing(MB0,0x00);

                                     ^

In file included from C:\Users\MANIVA~1\AppData\Local\Temp\arduino_modified_sketch_668396\mailbox_filtering_example_with_interrupts.ino:1:0:

C:\Users\manivannans\sketchbook\libraries\FlexCAN_T4-master/FlexCAN_T4.h:302:18: note: candidate: void FlexCAN_T4<_bus, _rxSize, _txSize>::setMBFilterProcessing(FLEXCAN_MAILBOX, uint32_t, uint32_t) [with CAN_DEV_TABLE _bus = (CAN_DEV_TABLE)1075642368u; FLEXCAN_RXQUEUE_TABLE _rxSize = (FLEXCAN_RXQUEUE_TABLE)256u; FLEXCAN_TXQUEUE_TABLE _txSize = (FLEXCAN_TXQUEUE_TABLE)16u; FLEXCAN_MAILBOX = FLEXCAN_MAILBOX; uint32_t = long unsigned int]

 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>

                  ^

C:\Users\manivannans\sketchbook\libraries\FlexCAN_T4-master/FlexCAN_T4.tpp:815:16: note: in expansion of macro 'FCTP_OPT'

 FCTP_FUNC void FCTP_OPT::setMBFilterProcessing(FLEXCAN_MAILBOX mb_num, uint32_t filter_id, uint32_t calculated_mask) {

                ^

C:\Users\manivannans\sketchbook\libraries\FlexCAN_T4-master/FlexCAN_T4.h:302:18: note:   candidate expects 3 arguments, 2 provided

 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>

                  ^

C:\Users\manivannans\sketchbook\libraries\FlexCAN_T4-master/FlexCAN_T4.tpp:815:16: note: in expansion of macro 'FCTP_OPT'

 FCTP_FUNC void FCTP_OPT::setMBFilterProcessing(FLEXCAN_MAILBOX mb_num, uint32_t filter_id, uint32_t calculated_mask) {

                ^

no matching function for call to 'FlexCAN_T4<(CAN_DEV_TABLE)1075642368u, (FLEXCAN_RXQUEUE_TABLE)256u, (FLEXCAN_TXQUEUE_TABLE)16u>::setMBFilterProcessing(FLEXCAN_MAILBOX, int)'

tonton81, 2021-12-17:
you need to post your code, clearly it says:
`
Can0.setMB(i,RX,EXT)
`
as an error, the remaining errors can just be a fallthrough of the initial error. I don't know or see how your code is layed out so i suggest you try one of the examples with filtering or post your code

Manivannan-maker, 2021-12-17:
The error occurs in default example code.  And I think it is related to line number 22?
mailbox_filtering_example_with_interrupts:22: error: within this context

#include <FlexCAN_T4.h>
FlexCAN_T4<CAN1, RX_SIZE_256, TX_SIZE_16> Can0;

#define NUM_TX_MAILBOXES 2
#define NUM_RX_MAILBOXES 6
void setup(void) {
  Serial.begin(115200); delay(400);
  Can0.begin();
  Can0.setBaudRate(250000);
  Can0.setMaxMB(NUM_TX_MAILBOXES + NUM_RX_MAILBOXES);
  for (int i = 0; i<NUM_RX_MAILBOXES; i++){
    Can0.setMB(i,RX,EXT);
  }
  for (int i = NUM_RX_MAILBOXES; i<(NUM_TX_MAILBOXES + NUM_RX_MAILBOXES); i++){
    Can0.setMB(i,TX,EXT);
  }
  Can0.setMBFilter(REJECT_ALL);
  Can0.enableMBInterrupts();
  Can0.onReceive(MB0,canSniff);
  Can0.onReceive(MB1,canSniff);
  Can0.onReceive(MB2,canSniff);
  Can0.setMBFilterProcessing(MB0,0x00,0xFF);
  Can0.setMBFilterProcessing(MB1,0x03,0xFF);
  Can0.setMBFilterProcessing(MB2,0x0B,0xFF);
  Can0.mailboxStatus();
}

void canSniff(const CAN_message_t &msg) {
  Serial.print("MB "); Serial.print(msg.mb);
  Serial.print("  OVERRUN: "); Serial.print(msg.flags.overrun);
  Serial.print("  LEN: "); Serial.print(msg.len);
  Serial.print(" EXT: "); Serial.print(msg.flags.extended);
  Serial.print(" TS: "); Serial.print(msg.timestamp);
  Serial.print(" ID: "); Serial.print(msg.id, HEX);
  Serial.print(" Buffer: ");
  for ( uint8_t i = 0; i < msg.len; i++ ) {
    Serial.print(msg.buf[i], HEX); Serial.print(" ");
  } Serial.println();
}

void loop() {
  Can0.events();
}

tonton81, 2021-12-17:
that example was submitted by a user, in any case the error shown is because "i" is not casted to the respective enum (MBx)

Manivannan-maker, 2021-12-18:
The issue is in line 22 in example code:  mailbox_filtering_example_with_interrupts
Even when I set the MB filter like below snippet , I am getting the same error .
  Can0.setMB(MB0,RX,EXT); The issue is in setMBFilterProcessing , when I replace the setMBFilterProcessing with setMBFilter I didnt get any error in compilation . Could you explain the difference between setMBFilterProcessing and setMBFilter. Because even in Readme file of FlexCAN you have mentioned about usage of setMBFilter only . Why we are including setMBFilterProcessing instead on setMBFilter in this example code : mailbox_filtering_example_with_interrupts

#include <FlexCAN_T4.h>
FlexCAN_T4<CAN1, RX_SIZE_256, TX_SIZE_16> Can0;

#define NUM_TX_MAILBOXES 2
#define NUM_RX_MAILBOXES 6
void setup(void) {
  Serial.begin(115200); delay(400);
  Can0.begin();
  Can0.setBaudRate(250000);
  Can0.setMaxMB(NUM_TX_MAILBOXES + NUM_RX_MAILBOXES);
  Can0.setMB(MB0,RX,EXT);
    Can0.setMB(MB1,RX,EXT);

  Can0.setMB(MB2,RX,EXT);

 /*
  for (int i = 0; i<NUM_RX_MAILBOXES; i++){
    Can0.setMB(i,RX,EXT);
  }
  for (int i = NUM_RX_MAILBOXES; i<(NUM_TX_MAILBOXES + NUM_RX_MAILBOXES); i++){
    Can0.setMB(i,TX,EXT);
  }
  */
  Can0.setMBFilter(REJECT_ALL);
  Can0.enableMBInterrupts();
  Can0.onReceive(MB0,canSniff);
  Can0.onReceive(MB1,canSniff);
  Can0.onReceive(MB2,canSniff);
  Can0.setMBFilterProcessing(MB0,0x00,0xFF);
  Can0.setMBFilterProcessing(MB1,0x03,0xFF);
  Can0.setMBFilterProcessing(MB2,0x0B,0xFF);
  Can0.mailboxStatus();
}

void canSniff(const CAN_message_t &msg) {
  Serial.print("MB "); Serial.print(msg.mb);
  Serial.print("  OVERRUN: "); Serial.print(msg.flags.overrun);
  Serial.print("  LEN: "); Serial.print(msg.len);
  Serial.print(" EXT: "); Serial.print(msg.flags.extended);
  Serial.print(" TS: "); Serial.print(msg.timestamp);
  Serial.print(" ID: "); Serial.print(msg.id, HEX);
  Serial.print(" Buffer: ");
  for ( uint8_t i = 0; i < msg.len; i++ ) {
    Serial.print(msg.buf[i], HEX); Serial.print(" ");
  } Serial.println();
}

void loop() {
  Can0.events();
}

tonton81, 2021-12-18:
`
The issue is in setMBFilterProcessing , when I replace the setMBFilterProcessing with setMBFilter I didnt get any error in compilation . 
`
This is a user submitted example that required a modified library, his patches ended up being included in setMBUserFilter, so that demo is outdated

closed 2021-12-20 by Manivannan-maker