tonton81/FlexCAN_T4 #36: mailbox_filtering_example_with_interrupts giving multiple compiler errors.

mjs513, 2021-12-19:
Hi Tony.  
This example I think is all messed up.  Could fix it but would just be guess on what he's doing or what he wants to do with that example.  These are the errors I am getting:
  Arduino: 1.8.18 (Windows 10), TD: 1.56-beta4, Board: "Teensy 4.1, Serial, 600 MHz, Faster, US English"
mailbox_filtering_example_with_interrupts: In function 'void setup()':

C:\Users\Merli\AppData\Local\Temp\arduino_modified_sketch_821650\mailbox_filtering_example_with_interrupts.ino:12:24: warning: invalid conversion from 'int' to 'FLEXCAN_MAILBOX' [-fpermissive]
     Can0.setMB(i,RX,EXT);
                        ^
In file included from C:\Users\Merli\AppData\Local\Temp\arduino_modified_sketch_821650\mailbox_filtering_example_with_interrupts.ino:1:0:

F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.h:284:18: note:   initializing argument 1 of 'bool FlexCAN_T4<_bus, _rxSize, _txSize>::setMB(const FLEXCAN_MAILBOX&, const FLEXCAN_RXTX&, const FLEXCAN_IDE&) [with CAN_DEV_TABLE _bus = (CAN_DEV_TABLE)1075642368u; FLEXCAN_RXQUEUE_TABLE _rxSize = (FLEXCAN_RXQUEUE_TABLE)256u; FLEXCAN_TXQUEUE_TABLE _txSize = (FLEXCAN_TXQUEUE_TABLE)16u; FLEXCAN_MAILBOX = FLEXCAN_MAILBOX; FLEXCAN_RXTX = FLEXCAN_RXTX; FLEXCAN_IDE = FLEXCAN_IDE]'

 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>
                 ^
F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.tpp:242:16: note: in expansion of macro 'FCTP_OPT'
 FCTP_FUNC bool FCTP_OPT::setMB(const FLEXCAN_MAILBOX &mb_num, const FLEXCAN_RXTX &mb_rx_tx, const FLEXCAN_IDE &ide) {
                ^
C:\Users\Merli\AppData\Local\Temp\arduino_modified_sketch_821650\mailbox_filtering_example_with_interrupts.ino:15:24: warning: invalid conversion from 'int' to 'FLEXCAN_MAILBOX' [-fpermissive]
     Can0.setMB(i,TX,EXT);
                        ^
In file included from C:\Users\Merli\AppData\Local\Temp\arduino_modified_sketch_821650\mailbox_filtering_example_with_interrupts.ino:1:0:

F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.h:284:18: note:   initializing argument 1 of 'bool FlexCAN_T4<_bus, _rxSize, _txSize>::setMB(const FLEXCAN_MAILBOX&, const FLEXCAN_RXTX&, const FLEXCAN_IDE&) [with CAN_DEV_TABLE _bus = (CAN_DEV_TABLE)1075642368u; FLEXCAN_RXQUEUE_TABLE _rxSize = (FLEXCAN_RXQUEUE_TABLE)256u; FLEXCAN_TXQUEUE_TABLE _txSize = (FLEXCAN_TXQUEUE_TABLE)16u; FLEXCAN_MAILBOX = FLEXCAN_MAILBOX; FLEXCAN_RXTX = FLEXCAN_RXTX; FLEXCAN_IDE = FLEXCAN_IDE]'
 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>
                  ^
F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.tpp:242:16: note: in expansion of macro 'FCTP_OPT'
 FCTP_FUNC bool FCTP_OPT::setMB(const FLEXCAN_MAILBOX &mb_num, const FLEXCAN_RXTX &mb_rx_tx, const FLEXCAN_IDE &ide) {
                ^
F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.h:284:18: error: 'void FlexCAN_T4<_bus, _rxSize, _txSize>::setMBFilterProcessing(FLEXCAN_MAILBOX, uint32_t, uint32_t) [with CAN_DEV_TABLE _bus = (CAN_DEV_TABLE)1075642368u; FLEXCAN_RXQUEUE_TABLE _rxSize = (FLEXCAN_RXQUEUE_TABLE)256u; FLEXCAN_TXQUEUE_TABLE _txSize = (FLEXCAN_TXQUEUE_TABLE)16u; FLEXCAN_MAILBOX = FLEXCAN_MAILBOX; uint32_t = long unsigned int]' is private

 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>
                  ^
F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.tpp:815:16: note: in expansion of macro 'FCTP_OPT'
 FCTP_FUNC void FCTP_OPT::setMBFilterProcessing(FLEXCAN_MAILBOX mb_num, uint32_t filter_id, uint32_t calculated_mask) {
                ^
mailbox_filtering_example_with_interrupts:22: error: within this context
   Can0.setMBFilterProcessing(MB0,0x00,0xFF);
                                              ^

In file included from C:\Users\Merli\AppData\Local\Temp\arduino_modified_sketch_821650\mailbox_filtering_example_with_interrupts.ino:1:0:
F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.h:284:18: error: 'void FlexCAN_T4<_bus, _rxSize, _txSize>::setMBFilterProcessing(FLEXCAN_MAILBOX, uint32_t, uint32_t) [with CAN_DEV_TABLE _bus = (CAN_DEV_TABLE)1075642368u; FLEXCAN_RXQUEUE_TABLE _rxSize = (FLEXCAN_RXQUEUE_TABLE)256u; FLEXCAN_TXQUEUE_TABLE _txSize = (FLEXCAN_TXQUEUE_TABLE)16u; FLEXCAN_MAILBOX = FLEXCAN_MAILBOX; uint32_t = long unsigned int]' is private
 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>
                  ^
F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.tpp:815:16: note: in expansion of macro 'FCTP_OPT'
 FCTP_FUNC void FCTP_OPT::setMBFilterProcessing(FLEXCAN_MAILBOX mb_num, uint32_t filter_id, uint32_t calculated_mask) {
                ^
mailbox_filtering_example_with_interrupts:23: error: within this context
   Can0.setMBFilterProcessing(MB1,0x03,0xFF);
                                              ^
In file included from C:\Users\Merli\AppData\Local\Temp\arduino_modified_sketch_821650\mailbox_filtering_example_with_interrupts.ino:1:0:

F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.h:284:18: error: 'void FlexCAN_T4<_bus, _rxSize, _txSize>::setMBFilterProcessing(FLEXCAN_MAILBOX, uint32_t, uint32_t) [with CAN_DEV_TABLE _bus = (CAN_DEV_TABLE)1075642368u; FLEXCAN_RXQUEUE_TABLE _rxSize = (FLEXCAN_RXQUEUE_TABLE)256u; FLEXCAN_TXQUEUE_TABLE _txSize = (FLEXCAN_TXQUEUE_TABLE)16u; FLEXCAN_MAILBOX = FLEXCAN_MAILBOX; uint32_t = long unsigned int]' is private
 #define FCTP_OPT FlexCAN_T4<_bus, _rxSize, _txSize>
                  ^
F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4/FlexCAN_T4.tpp:815:16: note: in expansion of macro 'FCTP_OPT'
 FCTP_FUNC void FCTP_OPT::setMBFilterProcessing(FLEXCAN_MAILBOX mb_num, uint32_t filter_id, uint32_t calculated_mask) {
                ^
mailbox_filtering_example_with_interrupts:24: error: within this context
   Can0.setMBFilterProcessing(MB2,0x0000000B,0xFF);
                                                 ^
Multiple libraries were found for "FlexCAN_T4.h"
 Used: F:\arduino-1.8.18\hardware\teensy\avr\libraries\FlexCAN_T4
 Not used: D:\Users\Merli\Documents\Arduino\libraries\FlexCAN_T4-master
within this context

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tonton81, 2021-12-19:
#33 
User submitted demo was integrated into current functions without moving *Processing from private since library uses it internally

tonton81, 2021-12-19:
Commit included functions for it:
`
    bool setMBUserFilter(FLEXCAN_MAILBOX mb_num, uint32_t id1, uint32_t mask);
    bool setMBUserFilter(FLEXCAN_MAILBOX mb_num, uint32_t id1, uint32_t id2, uint32_t mask);
    bool setMBUserFilter(FLEXCAN_MAILBOX mb_num, uint32_t id1, uint32_t id2, uint32_t id3, uint32_t mask);
    bool setMBUserFilter(FLEXCAN_MAILBOX mb_num, uint32_t id1, uint32_t id2, uint32_t id3, uint32_t id4, uint32_t mask);

`

mjs513, 2021-12-19:
Thanks Tony - that makes sense - all fixed now and issued a PR back to you.  Wow the library really changed.  Think PJRC is getting ready for a new release of TD.

closed 2021-12-19 by mjs513