tonton81/FlexCAN_T4 #45: Interrupt race condition on RX bubber

AzwarPrihastomo, 2022-01-31:
I found issue when using multiple can port to listen bus at high rate
i found that the way interrupt store data using this

FCTP_FUNC void FCTP_OPT::struct2queueRx(const CAN_message_t &msg) {
  uint8_t buf[sizeof(CAN_message_t)];
  memmove(buf, &msg, sizeof(msg));
  rxBuffer.push_back(buf, sizeof(CAN_message_t));
}

the problem is when we calling FCTP_FUNC uint64_t FCTP_OPT::events() this function is accessing rxBuffer may make race condition

so i'm propose to add guard when we access rxBuffer by disabling the interrupt
like below

FCTP_FUNC uint64_t FCTP_OPT::events() {
  if ( rxBuffer.size() ) {
    CAN_message_t frame;
    uint8_t buf[sizeof(CAN_message_t)];
    **NVIC_DISABLE_IRQ(nvicIrq);**
    rxBuffer.pop_front(buf, sizeof(CAN_message_t));
    **NVIC_ENABLE_IRQ(nvicIrq);**
    memmove(&frame, buf, sizeof(frame));
    mbCallbacks((FLEXCAN_MAILBOX)frame.mb, frame);
  }
  if ( txBuffer.size() ) {
    CAN_message_t frame;
    uint8_t buf[sizeof(CAN_message_t)];
    txBuffer.peek_front(buf, sizeof(CAN_message_t));
    memmove(&frame, buf, sizeof(frame));
    if ( write((FLEXCAN_MAILBOX)getFirstTxBox(), frame) ) txBuffer.pop_front();
  }
  return (uint64_t)(rxBuffer.size() << 12) | txBuffer.size();
}

AzwarPrihastomo, 2022-02-18:
the buffer that you use is not thread safe, so the head and tail index could be prone to race condition because of this interrupt

tonton81, 2022-02-18:
it's not thread safe, it's context safe

closed 2023-04-14 by tonton81