tonton81/FlexCAN_T4 #48: Calling write at 2Hz but it's sending messages at 3000Hz

jflorchi, 2022-06-09:
I'm running a slightly modified version of this: https://github.com/tonton81/FlexCAN_T4/blob/master/examples/CAN2.0_example_FIFO_with_interrupts/CAN2.0_example_FIFO_with_interrupts.ino

I have a simple loop that sends a message, delays 500 ms, and then loops. In theory it should be sending 2 messages per second, but while monitoring with an external device, it's sending over 3000 messages per second. I must be missing something simple.

tonton81, 2022-06-10:
if there are transceiver connection issues, termination issues, baudrate issues, or no nodes ACKing on the bus, thats the expected behaviour

jflorchi, 2022-06-10:
> if there are transceiver connection issues, termination issues, baudrate issues, or no nodes ACKing on the bus, thats the expected behaviour

That's it, thank you. My dumbass forgot the second termination resistor so it couldn't ACK.

closed 2022-06-10 by jflorchi