tonton81/FlexCAN_T4 #49: Can.write(msg) repeats sending same frame forever

Knio, 2022-06-21:
Hi, when writing a CAN message, everything is working, but the message gets re-sent in a loop forever, eventually filling up all the queues and blocking the whole bus. Is there some automatic re-sending code if a message is not ACKed? how can I clear the transmitter?

I'm using 2x Teensy 4.0 with a SN65HVD230 transceiver (one sending and one receiving, no other nodes, on a bench) and have confirmed everything is working with a probe.
I modified the example code slightly to send a unique byte counter for every byte and repeat the mailbox status to be able to debug better.

Receiver log: (same code but with the write commented out)
```
FIFO Enabled --> Interrupt Enabled
        FIFO Filters in use: 8
        Remaining Mailboxes: 8
                MB8 code: TX_INACTIVE
                MB9 code: TX_INACTIVE
                MB10 code: TX_INACTIVE
                MB11 code: TX_INACTIVE
                MB12 code: TX_INACTIVE
                MB13 code: TX_INACTIVE
                MB14 code: TX_INACTIVE
                MB15 code: TX_INACTIVE
MB 99  OVERRUN: 0  LEN: 8 EXT: 0 TS: 5249 ID: 76D Buffer: 0 1 2 3 4 5 6 7
MB 99  OVERRUN: 0  LEN: 8 EXT: 0 TS: 5383 ID: 76D Buffer: 0 1 2 3 4 5 6 7
MB 99  OVERRUN: 0  LEN: 8 EXT: 0 TS: 5517 ID: 76D Buffer: 0 1 2 3 4 5 6 7
MB 99  OVERRUN: 0  LEN: 8 EXT: 0 TS: 5651 ID: 76D Buffer: 0 1 2 3 4 5 6 7
MB 99  OVERRUN: 0  LEN: 8 EXT: 0 TS: 5785 ID: 76D Buffer: 0 1 2 3 4 5 6 7
MB 99  OVERRUN: 0  LEN: 8 EXT: 0 TS: 5919 ID: 76D Buffer: 0 1 2 3 4 5 6 7
MB 99  OVERRUN: 0  LEN: 8 EXT: 0 TS: 6053 ID: 76D Buffer: 0 1 2 3 4 5 6 7
MB 99  OVERRUN: 0  LEN: 8 EXT: 0 TS: 6187 ID: 76D Buffer: 0 1 2 3 4 5 6 7
(continues forever)
```

Sender log:
```
FIFO Enabled --> Interrupt Enabled
        FIFO Filters in use: 8
        Remaining Mailboxes: 8
                MB8 code: TX_INACTIVE
                MB9 code: TX_INACTIVE
                MB10 code: TX_INACTIVE
                MB11 code: TX_INACTIVE
                MB12 code: TX_INACTIVE
                MB13 code: TX_INACTIVE
                MB14 code: TX_INACTIVE
                MB15 code: TX_INACTIVE
FIFO Enabled --> Interrupt Enabled
        FIFO Filters in use: 8
        Remaining Mailboxes: 8
                MB8 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0x76D)(Payload: 0 1 2 3 4 5 6 7)
                MB9 code: TX_INACTIVE
                MB10 code: TX_INACTIVE
                MB11 code: TX_INACTIVE
                MB12 code: TX_INACTIVE
                MB13 code: TX_INACTIVE
                MB14 code: TX_INACTIVE
                MB15 code: TX_INACTIVE
FIFO Enabled --> Interrupt Enabled
        FIFO Filters in use: 8
        Remaining Mailboxes: 8
                MB8 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0x76D)(Payload: 0 1 2 3 4 5 6 7)
                MB9 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0x4BC)(Payload: 8 9 A B C D E F)
                MB10 code: TX_INACTIVE
                MB11 code: TX_INACTIVE
                MB12 code: TX_INACTIVE
                MB13 code: TX_INACTIVE
                MB14 code: TX_INACTIVE
                MB15 code: TX_INACTIVE
FIFO Enabled --> Interrupt Enabled
        FIFO Filters in use: 8
        Remaining Mailboxes: 8
                MB8 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0x76D)(Payload: 0 1 2 3 4 5 6 7)
                MB9 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0x4BC)(Payload: 8 9 A B C D E F)
                MB10 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0x489)(Payload: 10 11 12 13 14 15 16 17)
[... 116 lines cut ...]
  Serial.print(" EXT: "); Serial.print(msg.flags.extended);
  Serial.print(" TS: "); Serial.print(msg.timestamp);
  Serial.print(" ID: "); Serial.print(msg.id, HEX);
  Serial.print(" Buffer: ");
  for ( uint8_t i = 0; i < msg.len; i++ ) {
    Serial.print(msg.buf[i], HEX); Serial.print(" ");
  } Serial.println();
}

void loop() {
  Can0.events();
  static uint32_t byte_count = 0;
  static uint32_t timeout = millis();
  if ( millis() - timeout > 2000 ) {
    CAN_message_t msg;
    msg.id = random(0x1,0x7FE);
    for ( uint8_t i = 0; i < 8; i++ ) msg.buf[i] = byte_count++;
    Can0.write(msg);
    timeout = millis();
    Can0.mailboxStatus();
  }

}

```

msadie, 2022-06-21:
"Is there some automatic re-sending code if a message is not ACKed?"
Yes, this is specified by the CAN protocol and is handled in the teensy's
hardware CAN module. Every frame should be ACKd by every other node on the
bus, so until something starts ACKing, clearing the transmit buffer won't
help.
This sounds like a problem with your transceivers or bus termination. Do
you have 120 ohm termination at each end of the bus? Is the Rs pin of the
svn65hvd230 shorted to ground to select high speed mode?

msadie, 2022-06-21:
Maybe double check the connection from the receiving Teensy's Tx pin to the transceivers D pin.

tonton81, 2022-10-21:
no, auto retransmission is done by hardware not software. if you send one frame it will try forever until someone hets it. clearing the queues won't help if your code still submits a single transfer

closed 2023-04-14 by tonton81