tonton81/FlexCAN_T4 #50: About loop back(transmit and receive)

kay01-kwon, 2022-07-03:
Hello, I'm a graduate student at Yonsei university.

What I am doing is to transmit and receive CAN data.

However, whenever I try this, it seems that data does not transmit and the driver cannot receive data at the same time.

I use two CAN transceivers like this.

[image]

I hooked up pin 23, pin 22 to CRX1, CTX1 and pin 0, pin 1 to CRX2, CTX2 respectively. Also, I connected CAN_L to CAN_L and CAN_H to CAN_H.

You do not have to install any resistor because the transceiver already is equipped with resistors.

Here are some code.

#include <FlexCAN_T4.h>
#include <kinetis_flexcan.h>

FlexCAN_T4<CAN2, RX_SIZE_256, TX_SIZE_16> Can1;
FlexCAN_T4<CAN2, RX_SIZE_256, TX_SIZE_16> Can2;
static CAN_message_t msg;

void setup() {
  // put your setup code here, to run once:
  delay(1000);

  Can1.begin();
  Can1.setBaudRate(50000);
  Can2.begin();
  Can2.setBaudRate(50000);

  
  msg.id = 0x100;
  msg.len = 4;
  msg.buf[0] = 0x01;
  msg.buf[1] = 0x02;
  msg.buf[2] = 0x03;
  msg.buf[3] = 0x04;
}

void loop() {
  // put your main code here, to run repeatedly:
  CAN_message_t inMsg;
  while(Can2.read(inMsg))
  {
    Can2.read(inMsg);
    Serial.print("CAN bus 2: "); 
  }
  
  Can1.write(msg);

  Serial.print("Sending : ");
  for(int i = 0; i < msg.len; i++){
    Serial.print(msg.buf[i],HEX); Serial.print(" ");    
  }

  Serial.println();

  delay(1000);
}

What is the main problem? Can you recommend any thing?

tonton81, 2022-07-03:
why do you have 2 CAN2 busses in the constructor? it will not work this way

kay01-kwon, 2022-07-03:
I'm sorry. It is my mistake. Even though the constructor is modified, it cannot read msg data at all.

Previous one

FlexCAN_T4<CAN2, RX_SIZE_256, TX_SIZE_16> Can1;
FlexCAN_T4<CAN2, RX_SIZE_256, TX_SIZE_16> Can2;

Current one

FlexCAN_T4<CAN1, RX_SIZE_256, TX_SIZE_16> Can1;
FlexCAN_T4<CAN2, RX_SIZE_256, TX_SIZE_16> Can2;

There are two transceiver which share the two lines CAN_H, CAN_L.

tonton81, 2022-07-03:
check your terminations, 120 ohms both transceiver ends, and make sure your transceiver pin if any is pulled to normal state

kay01-kwon, 2022-07-05:
To Dear TonTon81,
The transceiver has a problem. When looking at the can low signal through oscilloscope, it does not show any difference while the can high signal works well. Even though two 120 ohm resistor are put to the two transceiver, there is any response to can low bus. By ordering just a chip, I will check it again. By the way, the can tx signal is weird compared to arduino due. If you want to request the picture in oscilloscope, let me notice it by your response. Thanks.
From Kay

kay01-kwon, 2022-07-12:
The problem is solved by replacing the transceiver.

closed 2022-07-12 by kay01-kwon