tonton81/FlexCAN_T4 #52: set up again CAN when communication is down

rgianluca00, 2022-09-06:
Hi! I am using Flexcan with a Teensy 4.1 to communicate with a motor. I would like to know if there is any way to reset the communication or to set it up again (with can.begin() maybe) when CAN fails (i.e the motor does not send its position anymore and i read only the last data sent). Thanks!!

tonton81, 2022-09-06:
there is not suppose to be a failure point when CAN is awake or asleep, the devices are suppose to resume from laat state as usual. There is no reason to turn off and on the controller  peripheral of the microcontroller, because it won't solve the issue you have with a motor and running code with a disabled peripheral will most likely hard lockup the mcu. you need to check the datasheet of this motor to get the commands to move or read position data

tonton81, 2023-04-14:
Teensy automatically resumes CAN operations when the network comes back, you should still manage your own request to your devices. If a network is down, the transmit boxes stay full and further transmissions are dropped until the current ones are cleared from transmit (when the network returns)

closed 2023-04-14 by tonton81