tonton81/FlexCAN_T4 #53: Flexcan_t4 write(msg) doesn't work

tmdwhsh, 2022-09-15:
I am using teensy 4.1 and teensyduino.
And connected two VESC at can 1 port.

Receving data from Two VESC's CAN message is normally working, but sending CAN message from teensy to VESC not working.

Sometimes work well, but sometimes can1.write(msg); return '-1' value.
when I checked CAN Bus with CAN analyzer, there was no CAN Message from teensy.

I checked back and front of 'can1.write(msg);' code, but msg.buf[0], msg.buf[1], msg.buf[2], msg.buf[3] has actual number data.

intermittently work well, or write function return -1.

How can i do?

tonton81, 2022-09-15:
you have an issue with termination or transceiver. check if your transceiver is not in listen only mode

tmdwhsh, 2022-09-15:
My code doesn't include setting about 34pin.

I think that Floating state of 34pin make switching standby and normal mode..right?

tonton81, 2022-09-15:
not sure i cant check right now, but in listen only mode you can read but not write. you said you are receiving but not sending so that may be an enable pin on that IC to activate it

ingjae, 2022-09-26:
I have same issue 
can read can data but can't write can data 
using below board and connect to teensy
[image]

is there any progress?
```
FIFO Disabled
	Mailboxes:
		MB0 code: RX_EMPTY	(Standard Frame)
		MB1 code: RX_EMPTY	(Standard Frame)
		MB2 code: RX_EMPTY	(Standard Frame)
		MB3 code: RX_EMPTY	(Standard Frame)
		MB4 code: RX_EMPTY	(Extended Frame)
		MB5 code: RX_EMPTY	(Extended Frame)
		MB6 code: RX_EMPTY	(Extended Frame)
		MB7 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB8 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB9 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB10 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB11 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB12 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB13 code: TX_INACTIVE
		MB14 code: TX_INACTIVE
		MB15 code: TX_INACTIVE
disable motor
result : 1
FIFO Disabled
	Mailboxes:
		MB0 code: RX_EMPTY	(Standard Frame)
		MB1 code: RX_EMPTY	(Standard Frame)
		MB2 code: RX_EMPTY	(Standard Frame)
		MB3 code: RX_EMPTY	(Standard Frame)
		MB4 code: RX_EMPTY	(Extended Frame)
		MB5 code: RX_EMPTY	(Extended Frame)
		MB6 code: RX_EMPTY	(Extended Frame)
		MB7 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB8 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB9 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB10 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB11 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB12 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB13 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB14 code: TX_INACTIVE
		MB15 code: TX_INACTIVE
disable motor
result : 1
FIFO Disabled
	Mailboxes:
		MB0 code: RX_EMPTY	(Standard Frame)
		MB1 code: RX_EMPTY	(Standard Frame)
		MB2 code: RX_EMPTY	(Standard Frame)
		MB3 code: RX_EMPTY	(Standard Frame)
		MB4 code: RX_EMPTY	(Extended Frame)
		MB5 code: RX_EMPTY	(Extended Frame)
		MB6 code: RX_EMPTY	(Extended Frame)
		MB7 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB8 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB9 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB10 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB11 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB12 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB13 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB14 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB15 code: TX_INACTIVE
disable motor
result : 1
FIFO Disabled
	Mailboxes:
		MB0 code: RX_EMPTY	(Standard Frame)
		MB1 code: RX_EMPTY	(Standard Frame)
		MB2 code: RX_EMPTY	(Standard Frame)
		MB3 code: RX_EMPTY	(Standard Frame)
		MB4 code: RX_EMPTY	(Extended Frame)
		MB5 code: RX_EMPTY	(Extended Frame)
		MB6 code: RX_EMPTY	(Extended Frame)
		MB7 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB8 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB9 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB10 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB11 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB12 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB13 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB14 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
		MB15 code: TX_DATA (Transmitting)(Standard Frame)(ID: 0xC2)(Payload: 2A 0 0 0 0 0 0 0)
```

tmdwhsh, 2022-09-26:
I solved the problem setting pin34 low(normal mode)

ingjae, 2022-10-27:
Actually I solved it by changing the board
sn65hvd230 to mcp2551 
[image]

then every thing is fine ...
 I heard that there are many bad copy boards of sn65hvd230

thank you .

tonton81, 2023-04-14:
All good! So yeah, anytime the transmit boxes remain full, always check transceiver and terminations. The way the controller is designed is once the message hits the transmit mailbox, the software no longer has control over the transmission, and the only culprit of it not sending out is anything attached after the GPIO pins :)

closed 2023-04-14 by tonton81