tonton81/FlexCAN_T4 #54: Extended CAN message not received

Labolivier49, 2022-10-13:
Hello,

I'am using the Flexcan_T4 library and everything is working well execpt that I cannot receive extended CAN ID. Standard frames are well received, and can be filtered out, but even without filter, I cannot receive frame with Extended ID.

[image]
 
In my example, 
STD frames 501 are well received, and frame 601 is well filtered out.
STD Frame 100 as well as EXT Frame with ID 0x18D373F2 are well sent...
However, the answer with frame ID 0x18D3F273 is well sent by my device but not received by the teensy with Flexcan, even with all filtereing removed...

it seems that someone else had the same issue: https://www.reddit.com/r/embedded/comments/b4dvbh/teensy_flexcan_library_not_receiving_extended_can/

regards

tonton81, 2022-10-13:
post your code and show me what mailboxStatus() displays

Labolivier49, 2022-10-13:
Hello,

Sorry for the wrong link...I will edit my post so people don't get confused

Here is my code. 
I am using a Teensy 4.0

`
#include <FlexCAN_T4.h>

FlexCAN_T4 <CAN3, RX_SIZE_256, TX_SIZE_16> CAN_3;

//*********DIAG***************************

uint8_t START_DIAG[] = {0x02, 0x10, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00};
uint8_t READ_DTC[] = {0x03, 0x19, 0x02, 0x0C, 0x00, 0x00, 0x00, 0x00};
uint8_t FLOW_CONTROL[] = {0x30, 0x00, 0x0A, 0x0C, 0x00, 0x00, 0x00, 0x00};

const uint32_t  DiagRespR = 0x18DAF273;
const uint32_t  DiagReqR = 0x18DA73F2;

const uint32_t  DiagReqL = 0x18DA72F2;
const uint32_t  DiagRespL = 0x18DAF272;

int clocktime = 0;
int state = 0;
int timeread = 0;
int intervalread = 60000;
int timeout = 10000;
int BytesTowait = 0;
bool Right = true;

//*********FRAME OUT***************************
const int FRAME_OUT_NUMBER = 6;
const int FRAME_OUT_DATA = 12;

//ID, Size, Period, TIMER, Data1, Data2......
int FRAMES_OUT[FRAME_OUT_NUMBER][FRAME_OUT_DATA]={
{0x300,3,20,0,0x00, 00, 0x00, 00, 00, 00, 00, 00},
{0x301,3,20,0,0x00, 00, 0x00, 00, 00, 00, 00, 00},
{0x420,8,20,0,00, 00, 00, 00, 0x00, 00, 00, 00},
{0x400,8,20,0,00, 00, 00, 00, 00, 00, 00, 00},
{0x100,8,20,0,00, 00, 00, 00, 00, 00, 00, 00},
{0x101,8,20,0,00, 00, 00, 00, 00, 00, 00, 00}
};

//*********FRAME IN***************************

const int FRAME_IN_NUMBER = 2;
int FRAME_IN[FRAME_IN_NUMBER]={0x500,0x501};

//**********************A MODIFIER***********************

  

void canSniff(const CAN_message_t &msg) 
{

  switch (msg.id) 
  {

    case DiagRespR:
    case DiagRespL:

      canSniffDiag(msg);
      break;
    
    case 0x500:
    
    for ( uint8_t i = 0; i < msg.len; i++ ) {
[... 255 lines cut ...]
    {
    Serial.print("0x");Serial.print(msg.buf[i], HEX); Serial.print(" ");
    } Serial.println();
    
    BytesTowait -= 7;
    if (BytesTowait <= 0)
    {
      Right = !Right;
      timeread = 100;
      state =0;
    }
    
    break;
    
  }
  
}

void copyDiag(uint8_t* src, uint8_t* dst, int len) {
    for (int i = 0; i < len; i++) {
        dst[i] = src[i];
    }
}`

[image]

Labolivier49, 2022-10-17:
Hello,

I have fixed the issue changing the setup() function.
I changed the CAN_3.setFIFOFilter(**0**, FRAME_IN[i], STD); and dispatched the frames to different filters (0,1,2,3...)
However, I though that one filter could accept up to 5 frames, but I must be mistaken somewhere..
Best regards

tonton81, 2022-10-17:
yes the first value is the index for FIFO, it's a little bit different than mailboxes as they each have one, FIFO has to go through an index for it's filters (0,1,2,3 etc)

philgert, 2022-11-07:
Hello, I also have issues receiving extended frames, not sure what the issue is. I have no problem receving or sending STD frames. No probelm to send EXT either. My setup is like bellow. 

  Can2.enableMBInterrupts();
  Can2.setMBFilter(REJECT_ALL);
  Can2.onReceive(MB0, callback1);
  Can2.setMBFilter(MB0, 0x705,);

  Can2.onReceive(MB1, callback2);
  Can2.setMBFilter(MB1, 0x75);

  Can2.onReceive(MB2, callback3);
  Can2.setMBFilter(MB2, 0xF2, 0xF3, 0xF6);
  Can2.enhanceFilter(MB2);

  Can2.onReceive(MB3, callback4);
  Can2.setMBFilter(MB3, 0x555);

  Can2.onReceive(MB4, callback5);
  Can2.setMBFilter(MB4, 0xE, 0xF);
  Can2.enhanceFilter(MB3);

  Can2.onReceive(MB5, callback6);
  Can2.setMBFilter(MB5, 0x90A);
  Can2.setMB(MB10, RX, EXT);

 Can2.mailboxStatus() gives "MB5 code: RX_OVERRUN"

Anyone have an idea what might be the issue?

Thanks in advance!

tonton81, 2022-11-07:
why you set a filter then configure mailbox woth setMB? That is backwards. Also check what mailboxStatus shows

philgert, 2022-11-07:
Thank you for your reply. 

I put the "Can2.setMB(MB10, RX, EXT);" Before the filter. 

mailboxStatus says that the MB is full. But my callback isnt triggered by the interrupt. 
MB5 code: RX_FULL

void callback6(const CAN_message_t &msg)
{
  Serial.println("new message");
  for (int i=0; i<8; i++) {
    Serial.println(msg.buf[i]);
  }
}