tonton81/FlexCAN_T4 #58: Issue with mailbox interrupts

samprt-heracles, 2022-12-09:
Hello,

I am trying to listen to 3 different extended CAN frames with IDs `0x0CF02984`, `0x0CF02983` and `0x0CF02985`.

Here is the code I am using in setup() :

``` can1.begin();
    can1.setBaudRate(250000);
    can1.setMaxMB(3);
    can1.setMB(MB0, RX, EXT); 
    can1.setMB(MB1, RX, EXT); 
    can1.setMB(MB2, RX, EXT);
    can1.enableMBInterrupts();
    can1.onReceive(MB0, interrupt_1);
    can1.onReceive(MB1, interrupt_2);
    can1.onReceive(MB2, interrupt_3);
    can1.setMBFilter(MB0, 0x0CF02984, EXT);
    can1.setMBFilter(MB1, 0x0CF02983, EXT);
    can1.setMBFilter(MB2, 0x0CF02985, EXT);
    can1.enhanceFilter(MB0);
    can1.enhanceFilter(MB1);
    can1.enhanceFilter(MB2);
    can1.mailboxStatus();
```

Here is the result of mailboxStatus() :
```
FIFO Disabled
	Mailboxes:
		MB0 code: RX_EMPTY	(Extended Frame)
		MB1 code: RX_EMPTY	(Extended Frame)
		MB2 code: RX_EMPTY	(Extended Frame)
```

The interrupt_1 and interrupt_2 are triggered as expected when a message is sent but the interrupt_3 function is not triggered using this code.
If I comment out everything related to Mailbox 1 and  Mailbox 2 however, interrupt_3 is triggered.

Any help understanding what I am doing wrong would be much appreciated !

tonton81, 2022-12-09:
try setting enableMBInterrupts last

samprt-heracles, 2022-12-13:
Sorry, that did not solve the problem.

samprt-heracles, 2022-12-13:
Filtering the frame with ID `0x0CF02985` (previously filtered in MB2) on a different mailbox seems to fix the issue.
I have reordered the filters to use different mailboxes than in my previous code and now I have the expected behavior.

I still wonder what could be causing this issue though.

samprt-heracles, 2023-02-08:
I have had the same issue with the CAN2 interface. I had to filter out 3 different frames on 3 different mailboxes which called 3 different interrupts . The interrupts only worked when the frame IDs where placed in the mailboxes in decreasing order:
- MB0 -> 0x18FFE122
- MB1 -> 0x18FEF200
- MB2 -> 0x18EFF922

Has anyone experienced the same behavior ? It is not really a problem as long as it is written somewhere I guess.

tonton81, 2023-04-14:
I ran code from your first post, and it works fine (at least on the github version tested today). The only thing is the filter for the MB, you don't designate EXT for it. The mailbox setMB is already set as EXT(ended), the Filter is implied to be extended and the EXT you put in the filter counts as a secondary ID (as if you did a dual ID filter). So simply remove EXT when setting a mailbox filter, a filter is automatically set based on the mailbox's EXT/STD state from setMB.

the only time EXT/STD is used in a filter is only when using the FIFO filter system.

tonton81, 2023-04-14:
```
MB 0  OVERRUN: 0  LEN: 8 EXT: 1 TS: 18148 ID: CF02984 Buffer: 4 4 4 4 4 4 4 4 

MB 1  OVERRUN: 0  LEN: 8 EXT: 1 TS: 18289 ID: CF02983 Buffer: 5 5 5 5 5 5 5 5 

MB 2  OVERRUN: 0  LEN: 8 EXT: 1 TS: 18430 ID: CF02985 Buffer: 6 6 6 6 6 6 6 6 

MB 0  OVERRUN: 0  LEN: 8 EXT: 1 TS: 53406 ID: CF02984 Buffer: 4 4 4 4 4 4 4 4 

MB 1  OVERRUN: 0  LEN: 8 EXT: 1 TS: 53548 ID: CF02983 Buffer: 5 5 5 5 5 5 5 5 

MB 2  OVERRUN: 0  LEN: 8 EXT: 1 TS: 53689 ID: CF02985 Buffer: 6 6 6 6 6 6 6 6 

MB 0  OVERRUN: 0  LEN: 8 EXT: 1 TS: 23170 ID: CF02984 Buffer: 4 4 4 4 4 4 4 4 

MB 1  OVERRUN: 0  LEN: 8 EXT: 1 TS: 23311 ID: CF02983 Buffer: 5 5 5 5 5 5 5 5 

MB 2  OVERRUN: 0  LEN: 8 EXT: 1 TS: 23451 ID: CF02985 Buffer: 6 6 6 6 6 6 6 6 

MB 0  OVERRUN: 0  LEN: 8 EXT: 1 TS: 58464 ID: CF02984 Buffer: 4 4 4 4 4 4 4 4 

MB 1  OVERRUN: 0  LEN: 8 EXT: 1 TS: 58606 ID: CF02983 Buffer: 5 5 5 5 5 5 5 5 

MB 2  OVERRUN: 0  LEN: 8 EXT: 1 TS: 58746 ID: CF02985 Buffer: 6 6 6 6 6 6 6 6 

MB 0  OVERRUN: 0  LEN: 8 EXT: 1 TS: 28648 ID: CF02984 Buffer: 4 4 4 4 4 4 4 4 

MB 1  OVERRUN: 0  LEN: 8 EXT: 1 TS: 28789 ID: CF02983 Buffer: 5 5 5 5 5 5 5 5 

MB 2  OVERRUN: 0  LEN: 8 EXT: 1 TS: 28929 ID: CF02985 Buffer: 6 6 6 6 6 6 6 6 

MB 0  OVERRUN: 0  LEN: 8 EXT: 1 TS: 63905 ID: CF02984 Buffer: 4 4 4 4 4 4 4 4 

MB 1  OVERRUN: 0  LEN: 8 EXT: 1 TS: 64046 ID: CF02983 Buffer: 5 5 5 5 5 5 5 5 

MB 2  OVERRUN: 0  LEN: 8 EXT: 1 TS: 64186 ID: CF02985 Buffer: 6 6 6 6 6 6 6 6 

MB 0  OVERRUN: 0  LEN: 8 EXT: 1 TS: 33669 ID: CF02984 Buffer: 4 4 4 4 4 4 4 4 

MB 1  OVERRUN: 0  LEN: 8 EXT: 1 TS: 33811 ID: CF02983 Buffer: 5 5 5 5 5 5 5 5 

MB 2  OVERRUN: 0  LEN: 8 EXT: 1 TS: 33952 ID: CF02985 Buffer: 6 6 6 6 6 6 6 6 

```

tonton81, 2023-04-14:
Also the ENUM value for EXT is 1, so essentially you were doing dual filter for example:
IDs: 0xCF02983 and 0x1
without the EXT in filter, it only allows 0xCF02983