tonton81/FlexCAN_T4 #59: Allow isoTP implementation per mailbox

david-res, 2023-01-03:
Is there any way that I can assign an isotp callback to a mailbox filter?
Similar to how I would assign a custom callback per mailbox?
Could this capability be developed?

I am making requests to multiple nodes, and while two of them always have the same ID and can retain the same set of callbacks/response code, one of them (the ECU) varies with up to six different response parsing options and would be easier for me to manage if I can isolate the response logic into separate callbacks per node, as at the moment everything is in one isotp callback, and I need to copy the entire logic between each ECU variant which is prone to mistakes and is taking up memory

tonton81, 2023-02-26:
per mailbox? not in it's current form, the fixed buffer is a collector of all mailboxes. it would require a rewrite of most of the code to do this

david-res, 2023-02-27:
Got it.
So ideally a smaller user defined iso tp buffer per MB would be ideal for me.
It just makes it easier code wise to reconfig the mb callback/buffer rather than having to maintain multiple versions of the isotp rx callback that contain handlers for multiple IDs