tonton81/FlexCAN_T4 #60: EXT ID only using 18 LSB of full 29 bit CAN ID

NicRichardson, 2023-01-25:
Hello,

During some development where I'm receiving CAN messages with extended IDs I found unexpected behaviour where a lot more messages were being allowed through the filter. I did some testing by sending a range of CAN messages with incrementing extended IDs into this mailbox and it appears that only the 18 LSBs of the extended ID is being used in the hardware filtering. 

I made a post on the PJRC forum discussing this: [Teensy 4.1 FlexCAN_t4 ext ID setMBFilter() 18 bits only?](https://forum.pjrc.com/threads/71998-Teensy-4-1-FlexCAN_t4-ext-ID-setMBFilter()-18-bits-only)

Looking at the library code it appears that the problem is using FLEXCAN_MB_ID_IDEXT() in setMBFilterProcessing() when setting FLEXCANb_MBn_ID() and creating the mask variable in setMBFilter(FLEXCAN_MAILBOX mb_num, uint32_t id1) when it shouldn't be used. The method is using FLEXCAN_MB_ID_IDEXT() which masks out the 18 LSB (0x0003FFFF) however this masks out the STD ID bits which double as the MSB of an EXT ID. 

I ran a test where I replaced FLEXCAN_MB_ID_IDEXT(filter_id) with (filter_id & 0x1FFFFFFF) and ran my test again and the filter didn't allow any but the message I wanted; it worked! I'll include a snippet of the two methods I modified to get this to work. 

```C++
FCTP_FUNC void FCTP_OPT::setMBFilterProcessing(FLEXCAN_MAILBOX mb_num, uint32_t filter_id, uint32_t calculated_mask) {
  bool frz_flag_negate = !(FLEXCANb_MCR(_bus) & FLEXCAN_MCR_FRZ_ACK);
  FLEXCAN_EnterFreezeMode();
  FLEXCANb_RXIMR(_bus, mb_num) = calculated_mask | ((FLEXCANb_CTRL2(_bus) & FLEXCAN_CTRL2_EACEN) ? (1UL << 30) : 0);
  FLEXCANb_MBn_ID(_bus, mb_num) = ((!(FLEXCANb_MBn_CS(_bus, mb_num) & FLEXCAN_MB_CS_IDE)) ? FLEXCAN_MB_ID_IDSTD(filter_id) : (filter_id & 0x1FFFFFFF));
  if ( frz_flag_negate ) FLEXCAN_ExitFreezeMode();
}

FCTP_FUNC bool FCTP_OPT::setMBFilter(FLEXCAN_MAILBOX mb_num, uint32_t id1) {
  if ( mb_num < mailboxOffset() || mb_num >= FLEXCANb_MAXMB_SIZE(_bus) ) return 0; /* mailbox not available */
  if ( (FLEXCAN_get_code(FLEXCANb_MBn_CS(_bus, mb_num)) >> 3) ) return 0; /* exit on TX mailbox */ 
  uint32_t mask = ( !(FLEXCANb_MBn_CS(_bus, mb_num) & FLEXCAN_MB_CS_IDE) ) ? FLEXCAN_MB_ID_IDSTD(((id1) ^ (id1)) ^ 0x7FF) : ((((id1) ^ (id1)) ^ 0x1FFFFFFF) & 0x1FFFFFFF);
  setMBFilterProcessing2(mb_num,id1,mask);
  filter_store(FLEXCAN_MULTI, mb_num, 1, id1, 0, 0, 0, 0);
  return 1;
}
```

Although this worked, I wasn't sure if this was a bug or a design decision due to some quirk with the HW. Additionally I'm not sure if this'll work in all cases including the range, multiple, or masking ID filtering, it worked for me and I see no reason why it wouldn't work elsewhere.

msadie, 2023-01-25:
I just wanted to note a typo in your code above, in case someone tries it. setMBFilter() calls setMBFilterProcessing2() instead of setMBFilterProcessing().

NicRichardson, 2023-01-26:
Oh thanks for pointing that out, both of those functions had 2 suffixed to separate my code from the original library code. I'll fix that typo.

tonton81, 2023-04-14:
for mailboxes i don't see an issue with this, but FIFO has capability for partial ID's due to a single 32bit filter entry register per filter that can support 2-4 partial IDs, each. Why would you use 4 partial extended IDs in a single register I still can't find a use, mightes well just filter from your callback for more precision or not use a filter at all :P 

tested this with/without enhancedFilters and nothing broke.

Although the header mapping made no sense when i glimpsed at it.
#define FLEXCAN_MB_ID_IDEXT(x)        (((x)&0x0003FFFF)<<0)
#define FLEXCAN_MB_ID_IDSTD(x)        (((x)&0x000007FF)<<18)

Fix: https://github.com/tonton81/FlexCAN_T4/commit/c7573b534d117fddc8252834d0e4cfa0725b835b

closed 2023-04-14 by tonton81