tonton81/FlexCAN_T4 #61: Different sample rate for arbitration and data phase

trollkno1l, 2023-02-26:
There are cases where the CAN arbitration for arbitration phase is not defined the same as for data phase.
In my case f.e. would be a sample rate for arbitration phase of 70 and sample rate for data phase of 80. 

As I found out you are not able to set different sample rates for arbitration and data phase by now. Is that correct?

CANFD_timings_t config;
config.clock = CLK_24MHz;
config.baudrate = 500000;
config.baudrateFD = 2000000;
config.propdelay = 190;
config.bus_length = 1;
config.sample = 80;
FD.setBaudRate(config);

tonton81, 2023-02-26:
Take a look at FlexCAN_T4FDTimings.tpp, specifically line 13.
It's a custom based multi selection rate based off of NXP timing calculations.

The CPU speed also affects the timings in FD mode. Leave the advanced argument as true to see all the timing possibilities

tonton81, 2023-02-26:
setBaudRate(CANFD_timings_t config, uint8_t nominal_choice, uint8_t flexdata_choice, FLEXCAN_RXTX listen_only, bool advanced)

config you already have
nominal and flexdata leave as 1, they are default suggested rates, listen only as true/false, and make advanced as true for a listing of all possible nominal/flexdata rates for your config settings. then, to try them, simply change the nominal/flexdata values in the arguments with the numbers from the console print

trollkno1l, 2023-03-02:
ok I understand...
In my usecase the CANoe config ist set to be as following:

arbitrationphase: bitrate 500k; samplerate 80%
dataphase:           bitrate 2000k; samplerate 70% 

...so config.sample can eather be 80 or 70...
I dont know how many cases there are where this is needed...but maybe it would be helpful to have classmembers like:

config.sampleArbit = 80; 
config.sampleData = 70;

Your libary is so super super awsome and full of features like ISO TP, which is so awsome...
Maybe thats what missing :D

tonton81, 2023-04-14:
The calculations were based off NXP's excel document configuration. Whatever is available in flexcan_t4 is only as capable as what they offered in the spreadsheet calculation. Unfortunately other than porting the calculations over and designing a console option listing for selecting different timings, I wouldn't be able to help in deeper baudrate configuration as it's beyond my current scope, that's one of the reasons flexcan_t4fd was delayed for awhile during beta, CANFD mode worked using fixed register values found on forums, before the baudrate calculator was implemented. So it wasn't viable for most people who had to search that time for specific rates