tonton81/FlexCAN_T4 #64: Reset does not reset FIFO

geffennir, 2023-04-09:
2 Teensy 4.1s connected via CANBUS.
Tx goes both ways at ~ 500 Hz.
Using polling mode
Works Fine until reset of 1 of them.
Recovery possible  if 
Can2.disableFIFO();
Can2.enableFIFO();

---
Can2.reset() does not solve recovery.

Init code:
Can2.begin();
Can2.setClock(CLK_60MHz);
if (u32BaudRate)
{
    Can2.setBaudRate(u32BaudRate);
}
Can2.enableFIFO();

Tx code:
message.id = 0xafebabe;
message.len = 8;
message.seq = 1;
message.flags.extended = 1;
bool bTxOk = Can2.write(message); // this line fails
// patch
if(bTxOk < 0)
{
   Can2.disableFIFO();
   Can2.enableFIFO();
}
// end patch
Rx code:
Can2.read(message);
TUInt32 u32Id;
if (message.flags.extended) // 29 bits available
{
    u32Id = message.id & 0x1fffffff;
}
else // only 11 bits available
{
    u32Id = message.id & 0x7ff;
}

// payload use with memcpy.

tonton81, 2023-04-14:
I tried polling, works here

are you using the correct termination?
can you try resetting the ESR before the returns and see if that helps?
```FLEXCANb_ESR1(_bus) = FLEXCANb_ESR1(_bus);```

geffennir, 2023-04-16:
Oh I wasn't clear on the Use Case.

This code works when I start both at the same time.
Afterwards I disconnect just 1 from power, and then reconnect it.
Other teensy needs a reset to FIFO (Can2.disableFIFO(); Can2.enableFIFO();) or the communication will not be restored.

I haven't dived in to driver code, just upper-level debug.

tonton81, 2023-04-16:
yes, thats why the ESR reset was implemented in the Interrupt version, does interrupt version work fine for you? if so, then we just need to test then add the ESR code to the polling functions