tonton81/FlexCAN_T4 #68: Unable to debug when using FlexCAN_T4 on platformIO

Moi78, 2023-10-19:
Hello,
I was trying to debug a program on a Teensy 4.0 but I found out that I was completely unable to compile my firmware in debug mode due to FlexCAN_T4.

Here is the error : 
```
c:/users/mathys.d/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\teensy40\src\main.cpp.o: in function `FlexCAN_T4<(CAN_DEV_TABLE)1075642368, (FLEXCAN_RXQUEUE_TABLE)256, (FLEXCAN_TXQUEUE_TABLE)16>::FlexCAN_T4()':
C:\Users\mathys.d\.platformio\packages\framework-arduinoteensy\libraries\FlexCAN_T4/FlexCAN_T4.tpp:68: undefined reference to `vtable for FlexCAN_T4_Base'
```
I am using PlatformIO in visual studio.

Thanks in advance,

ishelly404, 2024-08-08:
I ran into this issue as well, I was able to fix it by changing line 331 in FlexCAN_T4.h from 
`virtual uint8_t getFirstTxBoxSize();` to `virtual uint8_t getFirstTxBoxSize() = 0;`. The issue seems to be because of a missing definition of the virtual function `getFirstTxBoxSize()`.

I was able to debug without issue after making this change.

finhudson, 2024-10-28:
I'm having the same problem, though line 331 is already updated to what you said in the newest version. Anyone else got any ideas what might be causing this?

ZombieFodder, 2025-01-09:
I am also able to debug in PlatformIO using the change suggested by ishelly404. PlatformIO Teensy V5.0.0 had the same issues as OP suggested for me initially.