tonton81/FlexCAN_T4 #79: FlexCAN_T4 library not Sending ACK

sdevetis, 2024-08-12:
Hello.
I'm currently troubleshooting an issue with my system not receiving CAN messages. I have implemented the FlexCAN_T4 FIFO example, but I haven't succeeded. I've checked my oscilloscope and it shows that the ACK bit is staying at (1) when it should have been changed to (0) by the library. All I need is to be able to receive messages, not need to transmit messages. Please take a look at the attached pictures for more details.

The scope image captured the data at PIN-30 (RX). CAN messages are been received from the transceiver, but the Arduino-IDE/Serial Monitor doesn't display anything except this,

FIFO Enabled --> Interrupt Enabled
FIFO Filters in use: 8
Remaining Mailboxes: 8
MB8 code: TX_INACTIVE
MB9 code: TX_INACTIVE
MB10 code: TX_INACTIVE
MB11 code: TX_INACTIVE
MB12 code: TX_INACTIVE
MB13 code: TX_INACTIVE
MB14 code: TX_INACTIVE
MB15 code: TX_INACTIVE
Ready...

Hardware,
1 Teensy 4.1 NE, using CAN CRX3 (pin30), CTXR3 (pin31), (tried others Pins 0/1 and Pins 22/23 with same results)
1 SN65HVD230DR Transceiver with a 3.3-V supply and each end of the bus is terminated with a 120-Ω resistor in compliance with the standard to minimize signal reflections on the bus.

Sketch,
#include <FlexCAN_T4.h>
FlexCAN_T4<CAN3, RX_SIZE_256, TX_SIZE_16> myCan;
CAN_message_t msg;

void setup() {
Serial.begin(9600);
delay(400);

myCan.begin();
myCan.setBaudRate(1000000);
myCan.setMaxMB(16);
myCan.enableFIFO();
myCan.enableFIFOInterrupt();
myCan.onReceive(canSniff);

myCan.mailboxStatus();

Serial.println("Ready...");
}

void loop() {
myCan.events();
}

void canSniff(const CAN_message_t &msg) {

Serial.print("BUS: CAN"); Serial.print(msg.bus);
Serial.print(" MB "); Serial.print(msg.mb);
Serial.print(" OVERRUN: "); Serial.print(msg.flags.overrun);
Serial.print(" LEN: "); Serial.print(msg.len);
Serial.print(" EXT: "); Serial.print(msg.flags.extended);
Serial.print(" ID: "); Serial.print(msg.id, HEX);
Serial.print(" Data: ");
for (uint8_t i = 0; i < msg.len; i++) {
Serial.print(msg.buf, HEX);
Serial.print(" ");
}
Serial.print(" TS: "); Serial.print(msg.timestamp);
Serial.println();
}

[image]
[image]
[image]

msadie, 2024-08-13:
From what I can tell, it looks like you have normal pitfalls covered.
Any idea why the voltage on the rx pin is so low? Is this just due to how the scope is configured?
Have you scoped the tx pin (pin 31) to see if the teensy is trying to ack?

sdevetis, 2024-08-13:
Yes, it is due to how the scope is configured. I have not scoped the TX pin (Pin-31) for ACK, but I will check.

sdevetis, 2024-08-14:
Yes, the Teensy was doing the ACK, but the Transceiver was 1.2V on the RX side so I replaced it, so we are good to go,
Thank you for the help!!
[image]

closed 2024-08-14 by sdevetis