tonton81/FlexCAN_T4 #81: Can0.events()'s role

soonho-star, 2025-02-11:
I tested the <CAN2.0_example_FIFO_with_interrupts> example by including and removing Can0.events() in the main loop. Even without Can0.events(), the canSniff() function still worked properly when receiving CAN messages. So, is Can0.events() really necessary? What exactly is its(Can0.events()) role?

I read your explanation

tonton81, 2025-02-12:
events() is used when interrupts are not enabled, it may have been left in the example during testing, but it's also for non-interrupt enabled mailboxes, in which case both can be used together

soonho-star, 2025-02-13:
As I understand what you are saying, if the interrupt is used, the event function is not necessary, and if the interrupt is not used, it should be put in the loop.
Do I understand correctly?

tonton81, 2025-02-13:
yes, or you use it with interrupts if you want to use interrupt AND non interrupt mailboxes mixed, which would be weird but everyone's setups are different 😁

soonho-star, 2025-02-14:
Thank you for your kind reply!
I have some additional question...!
What does "non-interrupt mailbox" mean?
Are you referring to reading the mailbox (MB) using the readMB() function?
And I understand that the readMB() function reads all mailboxes, but is there a function that reads only a specific mailbox (e.g., a function to extract(or polling function) data from MB1)?